emmebrusa/TSDZ2-Smart-EBike-1 #3: Shell scripts for flashing with Unix systems?

mosscoder, 2020-05-22:
Greetings, thank you for putting this repository together. 

I'm a Unix system user and am interested in flashing the firmware of my TSDZ2. I have installed the dependencies that you describe in the linux/OSX install wiki and cloned this repository. 

I cannot locate the shell scripts described in the wiki, however. Is this still on the to do list? 

I have also tried to flash via the terminal after compiling in the configurator, however, I cannot locate the main.ihx after compilation.

Would you have any updates or guidance?

Sincerely,
Kyle

revher, 2020-05-22:
Hi, 
In the new OS/X Linux wiki page, I suppressed the use of the shell scripts because the advantage was very small in comparison to using the standard make command. Thus the Java configurator will help you to manage different settings (.ini files) and the compile and flash button will only create the config.h file. Then under src/controller you can do $ make -f Makefile_windows and you see that each .c is compiled into a .rel file and at the very end the main.rel is linked to other .rel to produce the main.ihx final file (which will be flashed). Don't use the Makefile_linux!! Ithink that it is explained in the wiki.

raboof, 2021-08-11:
> Don't use the Makefile_linux!! Ithink that it is explained in the wiki.

Is this still true? I'm not sure i understand - the wiki says:

> It also produces an .elf firmware file (--out-fmt-elf) instead of .ihx (--out-fmt-ihx) which is mainly used for debugging purposes

... but `Makefile_linux` seems to have a target to create the ihex:

https://github.com/emmebrusa/TSDZ2-Smart-EBike-1/blob/master/src/controller/Makefile_linux#L91

even though its `flash` target seems to flash the `.bin`(??):

https://github.com/emmebrusa/TSDZ2-Smart-EBike-1/blob/master/src/controller/Makefile_linux#L94

dzid26, 2022-08-28:
On linux I am getting
```
sdcc -c -I../common/STM8S_StdPeriph_Lib/inc -I../common -I. -I../ -mstm8 -Ddouble=float --std-c99 --nolospre  --out-fmt-ihx --debug  -oebike_app.c ebike_app.c
*** buffer overflow detected ***: terminated
Caught signal 6: SIGABRT
make: *** [Makefile_windows:84: ebike_app.rel] Error 1
```
Perhaps related to [this](https://sourceforge.net/p/sdcc/bugs/2741/), however, removing `--debug` doesn't help.

Probably `ebike_app.c` grew too much.

raboof, 2022-08-28:
> ```
> *** buffer overflow detected ***: terminated
> Caught signal 6: SIGABRT
> ```

What version of sdcc are you using? I *think* this sounds like the problem I proposed a fix for in https://sourceforge.net/p/sdcc/patches/397/ , which made it into 4.2.0 (but is not yet in 4.1.0)

dzid26, 2022-08-29:
> What version of sdcc are you using? 
@raboof Nice catch. 
I was using 3.8.0 from apt-get on Ubuntu 20.04.

It [worked](https://github.com/dzid26/TSDZ2-Smart-EBike/runs/8062156559?check_suite_focus=true) with 4.2.0 on [Ubuntu 22.04](https://sourceforge.net/p/sdcc/discussion/1865/thread/63e0e3e0eb/).