emmebrusa/TSDZ2-Smart-EBike-1 #5: Power down at 29v

Kati-Moa, 2020-07-12:
Mbrusa, the 36v FS did not switch off the display when battery reached 29v, and the odometer still indicated speed and distance whilst motor shut off. Given that we know the battery voltage rises when load is removed it will still be 'safe' to keep the display on so the distance traveled and speed is indicated. At present, the OSF shuts everything down and will not start up again. This is a disadvantage for the rider. The FS just flashes the first bar to indicate the battery is off. Perhaps if voltage once again falls to 29v the whole system should shut down as before.
Also there is a spelling error in Configurator. Assistant Setting page  'Street Power Limit enebled' should be  'Street Power Limit enabled'.
Ian

dzid26, 2022-12-02:
@emmebrusa did you experience the display shutting off when battery is low?
- Is [this bit](https://github.com/emmebrusa/TSDZ2-Smart-EBike-1/blob/51c5e2282bc3b1ebda94f53ca41c1ba2da81e429/src/controller/ebike_app.c#L3072) what triggers it?

- If the above is true, then I think the BATTERY_LOW_VOLTAGE_CUT_OFF  has to be set above BATTERY_CELLS_NUMBER * LI_ION_CELL_VOLTS_0_X100 to start derating the motor - before display shutdown. This 29V cut off in the documentation is a bit misleading, given that the default empty cell voltage is set to 3.1V now
[image]
So far I set the cell empty volt to 2.9V and cut off voltage to 41V (14 cells), but the display shut down anyway. I am going to increase cut-off voltage.

emmebrusa, 2022-12-03:
"Cell voltage empty (V)" 3.1V, is only used for SOC calculation compared to no load voltage.
"Battery voltage cut-off (V)" 2,9V * cell number, is used to decrease the power (down to zero) when the voltage under load reaches this value. It can be said that power is self-limiting.
The firmware does not turn off the display, if it happens, it is because the BMS of the battery intervenes.
Try increasing "Battery voltage cut-off (V)", so that it limits the power before the BMS intervenes.

dzid26, 2022-12-03:
Gosh, genius - it's got to be it. I will increase the cut-off voltage.

Regarding:
>"Cell voltage empty (V)" 3.1V, is only used for SOC calculation compared to no load voltage. 

When SOC value `ui8_battery_state_of_charge` is equal 0, the `ui8_working_status` bit 0 is set to 1. Do you know what that does on the display? I am wondering

emmebrusa, 2022-12-04:
I had tried, with VLCD5 it displays the low battery symbol even if it is full.
With XH18 nothing changes, it is probably not used.
I have not tried with other types of displays.

Kati-Moa, 2022-12-04:
Please don't send anymore emails as my husband has passed away and I don't
have the ebikes anymore.
Regards, Christine.

emmebrusa, 2022-12-04:
I'm sorry.
The sending of the email is automatic because her husband had started this discussion.
I close it and ask other users interested in the subject to open a new one.
Greetings.

closed 2022-12-04 by emmebrusa