emmebrusa/TSDZ2-Smart-EBike-1 #8: doesnt compile/flash when i click the button...

Adarsh881, 2020-09-18:
Hi. 
I did everything what was written in the instruction manual but in the end it doesnt compile/flash when i click the button.

I took a picture off the screen. sorry...printscreen is not working...
Thank you:)

[image]

eIinx, 2021-09-28:
SDCC and STVP must be installed inside the root for the bat-script that is reponsable for compile&flash

jay7019, 2022-09-02:
Adarsh881 did you figue this out im having same problem

Adarsh881, 2022-09-02:
Yes if I remember correctly you simply have to run it in windows administrator mode then it worked...

jay7019, 2022-09-02:
Thanks for your help