emmebrusa/TSDZ2-Smart-EBike-1 #11: Unable to compile last release (v20.1C.3 8a0872e) on OSX

espenfe, 2021-06-09:
I'm unable to compile the latest release (v20.1C.3 8a0872e) on OSX.

The compilation fails on one of the last files (ebike_app.c) with the following error message:
sdcc -c -I../common/STM8S_StdPeriph_Lib/inc -I../common -I. -I../ -mstm8 -Ddouble=float --std-c99 --nolospre  --out-fmt-ihx --debug  -oebike_app.c ebike_app.c
**make: *** [ebike_app.rel] Trace/BPT trap: 5**

The system is built with SDCC 4.1.0 with make -f Makefile_windows

Reverting to the previous commit compiles fine - no problem.

emmebrusa, 2021-06-17:
v20.1C.3 is suspended due to potential risk to mosfets.

apolselli, 2021-07-04:
Same compilation error on linux for v20.1C.1

Mikwind, 2021-07-18:
I get a slightly different error on v20.1c.3 NEW  6891be19 under linux, V20.1c.1 also gives the same error. v20.1C compiles fine. using sdcc 3.8.0:

Makefile_windows:84: update target 'ebike_app.rel' due to: ebike_app.c torque_sensor.h interrupts.h main.h uart.h pwm.h motor.h wheel_speed_sensor.h brake.h pas.h adc.h timers.h ebike_app.h pins.h eeprom.h lights.h config.h
sdcc -c -I../common/STM8S_StdPeriph_Lib/inc -I../common -I. -I../ -mstm8 -Ddouble=float --std-c99 --nolospre  --out-fmt-ihx --debug  -oebike_app.c ebike_app.c
*** buffer overflow detected ***: terminated
Caught signal 6: SIGABRT
make: *** [Makefile_windows:84: ebike_app.rel] Error 1

apolselli, 2021-07-29:
To solve the *** buffer overflow detected *** problem on linux, caused by the sdcc package that is compiled with -D_FORTIFY_SOURCE=2 flag by ubuntu/debian mainiainers (https://wiki.ubuntu.com/ToolChain/CompilerFlags),
you can follow these instructions (for ubuntu/amd64):
```
# remove the sdcc package
sudo apt remove sdcc
sudo apt autoremove
```
```
# install the latest sdcc snapshot build
wget http://sourceforge.net/projects/sdcc/files/snapshot_builds/amd64-unknown-linux2.5/sdcc-snapshot-amd64-unknown-linux2.5-20210711-12539.tar.bz2
tar xjf sdcc-snapshot-amd64-unknown-linux2.5-20210711-12539.tar.bz2 
cd sdcc
sudo cp -r * /usr/local
```

emmebrusa, 2021-08-03:
I tried to compile in a Linux environment.
Now works!
Thank you

espenfe, 2021-08-06:
This does not fix the issue for compilation on OSX however - I still get the same compilation error.

Removing D_FORTIFY_SOURCE=2 removes code hardening checks I guess so one should look for writing past memory regions etc in ebike_app.c

apolselli, 2021-08-06:
D_FORTIFY_SOURCE=2 was used compiling the compiler (sdcc), not this code. The (possible) buffer overflow detected at runtime is in the sdcc code, not here

raboof, 2021-08-10:
> *** buffer overflow detected ***: terminated
> Caught signal 6: SIGABRT
> make: *** [Makefile_windows:84: ebike_app.rel] Error 1

I posted a patch to sdcc to make it work with sdcc again [here](https://sourceforge.net/p/sdcc/patches/397/). It's not clear whether that'd also fix the OSX problem that this thread started with.

> **Nevermind, updating to SDCC 4.1 fixed it**

(the problem is still there on sdcc SVN HEAD, so I suspect your SDCC 4.1 was just compiled without fortify)

espenfe, 2021-08-20:
I looked at this today again and read that your patch raboof were included in one of the latest commits to sdcc! 

Downloading the latest snapshot build of sdcc for OSX and installing it worked for me! I'm now compiling v20.1C.3 on OSX

This will complicate set up of the build environments on OSX and linux for.a while until SDCC comes with a new official release... Maybe there should be a cookbook how-to for both Linux and OSX available? For Linux one can follow apolsellis how-to and for OSX its as follows:

remove currently installed version of SDCC - mine was installed using homebrew so brew uninstall sdcc

And then install sdcc from snapshot: 
download latest snapshot from https://sourceforge.net/projects/sdcc/files/snapshot_builds/x86_64-apple-macosx/
tar xjf sdcc-snapshot-xx 
cd sdcc
sudo cp -r * /usr/local

closed 2021-08-20 by espenfe