emmebrusa/TSDZ2-Smart-EBike-1 #12: Battery soc wh over 900

dudarus1, 2021-06-12:
I have a 28ah battery but I can't get the wh any higher than 900 thanks in advance jake

emmebrusa, 2021-07-02:
In the JavaConfigurator, "Battery capacity (Wh)" can be set to any value and the remaining percentage calculation should be correct.
If you are referring to the "Used Wh" consumption display this is limited to 99.9 (999Wh or even less depending on the display settings), because at 1000Wh I would have to change the scale and display 10.0, it would only create confusion.

dudarus1, 2021-07-02:
It's the percentage that I'm referring to

I'm using 850c display and in soc menu the wh maxes out at 900wh

Kind regards
Jake

emmebrusa, 2021-07-02:
You asked the question in the stock display version, for that I did not understand.
I have seen, in the version for 850C / 860C there is the max limit at 999.0
I will change it in the next version.

dudarus1, 2021-07-03:
Sorry for the confusion

Thanks for the assistance

Kind regards
Jake