emmebrusa/TSDZ2-Smart-EBike-1 #13: Thump throttle at 6 km/h in street mode

CableCatDK, 2021-07-10:
Instead the disabling the thump throttle in street mode, I would prefer it was set at 6 km/h.

karolgoluch, 2021-08-10:
I think street mode is to make "legal" ebike, so throttle in street mode with specified speed should be only an option.
On the other hand I would prefere prefer throttle still enabled when the assistance is off in off-road mode.

CableCatDK, 2021-08-12:
Just to clarify:  Thump throttle up to 6 km/h is legal in the EU.

karolgoluch, 2021-08-12:
Well, I didn't want to discuss here about law in Poland, but here it is https://isap.sejm.gov.pl/isap.nsf/download.xsp/WDU19970980602/U/D19970602Lj.pdf [Art. 2, 47), page 9 ]

"The bicycle may be equipped with a runnable with pressure on pedals electric drive".

This leaves no doubt that you cannot use throttle in electric bicycle in Poland, as electric drive is allowed only to run with pressure on pedals.