emmebrusa/TSDZ2-Smart-EBike-1 #16: Possible TSDZ2 destroyed while using v20.1C.3-NEW firmware

CableCatDK, 2021-07-29:
One of my TSDZ2 stopped working:
* If it is powered on, the power is immediately cut by the battery.
* If I rotate the bike backward, there is a mechanical sound.

I will investigate it further.

The motor is 5 month old and have ran 340 km.

The breakdown happed while carry a heavy load.

It was using this firmware: v20.1C.3-NEW

With these settings:
[image]

emmebrusa, 2021-07-30:
The settings are correct.
If the power is disconnected immediately upon power up, there is at least one short-circuited mosfet.
Moving the bike backwards you should feel resistance, if you also hear a mechanical noise there is something else to check.

CableCatDK, 2021-08-03:
It was a broken MOSFET. Good call :-)
This is most likely completely unrelated to the firmware.

TreadMTB, 2021-10-30:
> It was a broken MOSFET. Good call :-) This is most likely completely unrelated to the firmware.

is this the reason this version was suspended? what was your solution cablecatdk? did you need a brand new controller?

CableCatDK, 2021-10-30:
A new controller fixed the issue. I am running the same firmware on two other motors since the firmware was released. The only issue I noticed is that motor sometimes crashes, and need to be rebooted. In the crashed state the motor is halted, and the display is frozen.