emmebrusa/TSDZ2-Smart-EBike-1 #18: Parameter Configuration confusion

karolgoluch, 2021-08-09:
Can the parameters of "Battery current max" and "Battery power max" be explained more? What parameter is used with priority and when?

[image]
[image]

**Example 1:**
If I set:
Battery current max: 10A
Battery power max: 750W

and the motor is 48V, what will be max current?

P=U*I => I=P/U
I=750W/48V=15,625A

Will the current be limited to 10A or raised to 15,625A?

**Example 2:**
Battery current max: 18A
Battery power max: 250W

Will the current be limited to 5,2A (250W/48V)? Or will the power be raised to 864W (18A*48V)?

Or should this be calculated:
if: Battery current max = 12A
then: Battery power max = 576W

emmebrusa, 2021-08-20:
Between the two parameters, the priority is always the lower value.
Example 1: current limit 10A, the power limit changes with voltage, 52V * 10A = 520W, 40V * 10A = 400W.
Example 2: power limit 250W, current limit changes with voltage, 250W / 52V = 4.8A, 250W / 40V = 6.25A.

My advice is to set the parameters in order to have a constant maximum power, independent of the battery charge.
Example:
Battery power max = 520W (current 520W / 52V = 10A)
Battery current max = 13A (power 13A * 40V = 520W)
Attention the maximum power of 520W cannot be maintained continuously!

mja43, 2021-09-22:
Haloo,
i hawe confusion between parameeters „Number of data displayed at lights on“ and „Auto display data with lights on“ 
I have seted:
-	 „Number of data displayed at lights on“ = 1
-	 Data1 is set to 0 – motor temperature
-	„Auto display data with lights on“ – on
-	data 1- 6 all set to numbers from 0 to 8
-	time to display data 1-6 to 50
When i switch lights on i can only see only data 1 = motor temperature, other values dont showed after 5s interval.
Shud i let parameter „Number of data displayed at lights on“ blank to see data 1to6 in 5sec sekvence?

mk2-luke, 2021-09-22:
You should set number of data displayed at lights on to the number of data
fields you are using from the 6 available, set to 1 it will only show the
first field

mja43, 2021-09-22:
Thank you for explain, so this is a data limiter - show data numbers values from data1 to this data number value, not exactly data number to display