emmebrusa/TSDZ2-Smart-EBike-1 #21: v20.1C.3 suspended version motor running on its own without me pedaling

TreadMTB, 2021-10-30:
I was using the v20.1C.3 new suspended version because I was unaware of why it was suspended at the time. In my time testing it, I found that the motor engaging on its own may be related to using the assist without sensor mode. I turned the mode on to get it working when my speed sensor was missing, instead it started sporadically twitching forward into the wall, I held the brakes but I have no brake sensors as you don't generally need them with good torque sensing. It kept twitching till I turned it off and on. I was wondering if it was because I installed the temperature sensor that replaced the throttle wires, and the temperature reading is sending throttle signal through the wires in sensor less mode? I have temperature sensor enabled, so I would imagine that is not the cause. This was the only time it happened. I will go back to using 1c.1 for now, is their a particular or major reason the 20.1c.3 was suspended? thanks again for all the hard work

emmebrusa, 2021-11-21:
The versions v20.1C.3 for stock displays, is suspended because there have been cases of mosfet burn. I thought I had found a bug as a potential cause and I released v20.1C.3-NEW, but this version also had the same problems.
This is why it has been suspended again.

The motor running without pedaling is another problem.
It is rare and is also present in all previous versions of OSF for stock displays.
Also for this problem I thought I had found the solution, unfortunately it is not so, the problem is still there. I hope to find the cause soon.

karolgoluch, 2021-12-30:
Is there any "safe" version available now? I've actually burned two controllers so far.

emmebrusa, 2021-12-30:
But which version did you burn the controllers with?

karolgoluch, 2021-12-30:
20.1C

emmebrusa, 2021-12-30:
In addition to v20.1C.3 which has been suspended, if you go to the releases page:
https://github.com/emmebrusa/TSDZ2-Smart-EBike-1/releases
there is also a recommendation not to use v20.1C
"Warning, do not use this version.
Has been replaced by v20.1C.1 safety updates. "
Then there are two other versions:
v20.1B - is secure but does not have all the features of later versions.
v20.1C.1 - it is quite secure.
in both there is the rare problem of the motor turning without pedaling.
If this happens, with v20.1B just turn off the display.
In the case of v20.1C.1, turning off the display with the motor running can cause problems with the controller, better disconnect the battery.

woody100DK, 2022-02-05:
Is there still an issue with the motor running uncontrollably? That is quite worrying. How often does that happen and is there a version where this definitely doesn't happen?

woody100DK, 2022-02-05:
Is there still an issue with the motor turning uncontrollably? How often does that happen and is there a version where it definitely doesn't happen? When you say the motor turns, is it full speed or anything dangerous?

darkmattercoder, 2022-04-08:
for me it is supporting at full power, and happens absolutely random. Maybe at about every third ride or so? I simply turn the display off, as I cannot disconnect the battery during  a ride.