emmebrusa/TSDZ2-Smart-EBike-1 #22: How to increase the power of my 250W TSDZ2 to 500W

Oricb, 2021-11-28:
Congratulations for the firmware TSDZ2-OSF-V20.1C.3. I have installed on my bike with the TSDZ2 motor and I have found it very good. Thanks for sharing it.
But testing my bike, I have realised that the maximum power that it gives is 240W. I have a 36V battery, 630WH. With all the assit types, the bike does not increase the 240W. MAximum 6,6Amp when the battery was full (40V) and 7,5Amp with the battery quite empty. Making up to 200W of human power, in all assist modes and all assist levels, always maximum 240W of motor output.
**I have read that any motor can give more than the nominal output power, and the limitation of the maximum power was the max Current allowed. Is that true? Or I have to buy a new motor of 500W?**
The set up parameters that I have are your recommended values. But the 36V battery.
TSDZ2 w/860C
OSF Upgrade V20.1C.3

Battery
Max current	15 Amp
Low cut-off	30 volts
Resistance	200 milliohms

SOC
Text	Volts
Reset at voltage / Reset at	41.0
Battery total Watt-hour / Battery total	600
Used Watt-hour	0
Manual reset	no

Motor
Motor voltage	36V
Max motor power	450W
Motor Acceleration	35
Motor Deceleration	0
Motor fast stop	no
Field weakening	enabled

Torque sensor 
Assist w/o pedal rotation	Enable
Torque s ADC threshold 	10
Coast brake 	Enable
Coast brake ADC 	10
Calibration	Enable
Torque sensor ADC step	67
Torque sensor ADC offset	150
Torque ADC max	300
Weight on pedal	25 kg
Torque ADC on weight	250
Default weight	no

Assist level / Assist
Number of assist levels / Num levels	9	
Power Assist	9	Default
Torque Assist	9	Default
Cadence Assist	9	Default
eMTB assist	9	Default

Thanks for your answer.
Oriol Barcelona

emmebrusa, 2021-12-07:
OSF version 860C V20.1C.3 is another repository.
You are probably in Street mode with a power limit of 250W.
Check in the configuration menu, "Street mode" item.

Oricb, 2021-12-09:
Many thanks!! I am now in a travel, and when I come back I will try it!Thanks again.Best regards Oriol

Oricb, 2021-12-15:
Thanks a lot!

You were right. Street mode was enable, with a power set of 250W. Now, the power is limited with the parameter max power of the settings.

 

One more question if you allow me: I have read that the firmware is suspended because some mosfets have been blown up. I suppose that this is because people go with full power above 450W for too long. If I limit power to 450W, and I don’t ride at this power for more than some seconds, probably the mosfet won’t have any problem. What do you think?

 

I find the firmware incredible! The behaviour of the bike is much better than the standard firmware. You’ve done an incredible job. Many thanks for sharing!

 

Best regards

 

Oriol

Enviado el: martes, 7 de diciembre de 2021 18:06
Para: emmebrusa/TSDZ2-Smart-EBike-1 ***@***.***>
CC: Oricb ***@***.***>; Author ***@***.***>

emmebrusa, 2021-12-15:
The suspended version due to mosfet problems is only the one for stock displays.
No problems have been reported with the version for 860C.
I have three bikes with 36V motor and I have limited the power to 500W, it is fine for a few minutes.
I have the temperature sensor installed, it rarely intervenes to reduce the power.

Oricb, 2021-12-16:
> Enviado el: miércoles, 15 de diciembre de 2021 20:24

Many thanks for the information!

 

Best regards

 

Oriol