emmebrusa/TSDZ2-Smart-EBike-1 #26: Compatible with 'new' (late 2021) TSDZ2?

clort81, 2022-05-14:
Some forum posts have indicated no OSF works with 'new' TSDZ2 controllers/motors.  The forum posts were from 2021.

Is that correct?  Has this firmware been tested with recent (2021-2022) TSDZ2 motors?

Thanks

attixray, 2022-05-17:
I have bough my TSDZ2 1 month ago and it works perfectly. (Although, I have not checked the manufacturing date.)

jeremielec, 2022-05-26:
Same here flashed yesterday and working fine.

Note for me, i have to use the 3.3v pin as power source and not the 5V as mentionned in some documentation.

j0bro, 2022-09-24:
I have bought my 3rd TSDZ2 from PSWPower - which is now a TSDZ2B - and I can confirm it works with this firmware.

Ornias1993, 2022-10-14:
Important note:
There are basically 3 versions now:
- TSDZ2 "v1": old controller, old design
- TSDZ2 "v2": New controler, old design
- TSDZ2B: Old controller, changed design

None of the opensource firmware are compatible with TSDZ2 "v2", but most are compatible with both TSDZ2 "v1" and TSDZ2B