emmebrusa/TSDZ2-Smart-EBike-1 #30: getting error no matches found: stm8s105?6 when testing flash

mrulke, 2022-07-31:
everything seems to be going fine with the install doco but when running the test flash to backup the firmware I get the below

`running controller % stm8flash -c stlinkv2 -p stm8s105?6 -r TSDZ2_orig.bin`

not sure what that means I checked the wires and the stmv2 is lit up.

Dizotoff, 2022-11-20:
I think I've figured it out, just use quotation marks around `stm8s105?6` like: 
 
`stm8flash -c stlinkv2 -p "stm8s105?6" -r TSDZ2_orig.bin`

Woobieman, 2023-03-02:
I am seeing this same issue on MacOS 10.15 Catalina, quoting the parameter does not seem to work in zsh.

cravl-dev, 2023-05-18:
I'm also using ZSH (on KDE Neon), and I had to escape the question mark with a backslash, no quotes needed even. I also needed sudo, of course.

`sudo stm8flash -c stlinkv2 -p stm8s105\?6 -r TSDZ2_orig.bin`

dzid26, 2023-05-18:
> I'm also using ZSH (on KDE Neon), and I had to escape the question mark with a backslash, no quotes needed even. I also needed sudo, of course.
> sudo stm8flash -c stlinkv2 -p stm8s105\?6 -r TSDZ2_orig.bin

Is `\` needed inside makefile too?