emmebrusa/TSDZ2-Smart-EBike-1 #36: 36V coaster brake bike overrun problem

jussisaily, 2022-10-03:
I built a single speed bike using TSDZ2 36V coaster brake model and flashed with newest emmebrusa firmware. The bike works ok otherwise but there is a slight overrun felt in the pedals when I stop pedaling (not yet pushing backwards for coaster braking). It's like a forward yank on the pedals. Probably motor inertia before detecting no pedaling action. I was using hybrid mode.

I tried to increase the motor deceleration setting to 100% for fast stop but it makes no difference.

Is there any parameter or code to change to minimize the annoying overrun?

emmebrusa, 2022-10-04:
Try to set "Pedal torque ADC offset adjustment" with a positive value, maximum 14.
Or try Power mode with Startup boost disabled.

jussisaily, 2022-10-04:
Thanks, I will try those settings. The forward pedal 'yank' or 'overrun' is only felt when the bike is moving and I stop pedaling (not braking).

jussisaily, 2023-05-18:
I finally got back to this bike after the long winter... And tried the suggested parameter changes to the torque ADC threshold from 0 to 14 but there is really no difference in the behaviour. Also tried to change the coaster brake threshold values on the 3rd setting page but also no effect. I have 100% motor deceleration setting also.

There are multiple mentions in internet forums about the coaster brake versions problem with pedal yank in both original firmware and OSF. This is believed to be caused by the different drive mechanics, as the coaster brake variant has the motor directly on the pedal shaft without a sprag clutch in the regular version which probably smoothens out the yank. Someone mentioned that he has installed a brake sensor to the hand brake and pressing it before stopping pedaling stops the motor from yanking. 

I guess the problem in this (and original) firmware with coaster brake is that the torque sensor based pedaling sensing is too slow to stop the motor assist before a slight yank is felt in the pedals. Based on my testing the yank is bigger if I use the maximum assist level. It's less noticeable with power assist mode, and very noticeable in hybrid assist mode. Hybrid mode uses the torque sensing so it's slower to turn off, I understand.

I'm sure that users will get used to this, but it would be great if there was a way to incorporate faster motor turn-off without external brake sensors!

dzid26, 2023-05-18:
I had some thoughts about inertia because I noticed that when climbing a hill if I reduce pedal force I still feel forward torque from the motor. And that is more than on the flat. But in the end I concluded that this is simply due to using higher currents when climbing in general and simply a saturation of the assist level current due to bemf. It is more pronounced at higher assist levels, because the requested current is then often above what is physically possible. 

But that is probably not your case, because you mentioned a yank which implies short surge forward. Mine was longer. Back to the motor inertia then! I happened to do back of a napkin inertia torque calculation today, so there it is:

Stator + rotor weight `800g`, so let's say the rotor weight `300g` and assume it's `4cm` wide. Also assuming it is a [solid cylinder](https://en.wikipedia.org/wiki/List_of_moments_of_inertia), the moment of inertia of the stator is `1/2x0.3x0.02^2= 0.00006kg` m^2. The gearbox ratio is `41.8:1`.  Any acceleration that motors see is 42x the cadence acceleration. Any torque that pedals see, is 42x the motor torque (including inertial torque). Thus the equivalence of the moment of inertia from the motor converted to as if it was at the pedals has to be squared.  So `0.00006x41.8^2 = 0.1048344kgm2`.   Converting that even further to as if this mass was attached to the pedals `0.1048344 x (100cm/17cm)^2 = 3.62kg`. Additionally, there is a heavy 581g gear wheel, but it maybe will add like effective ~200g of mass converted to the pedals domain. Of course, this is "rotational mass" so it is not affected by gravity but by acceleration and deceleration. But you will feel those "3.8kg" pedals as they continue to "go forward" a bit when you try to stop them rapidly.

On top of that, there is a **motor torque reduction delay**. It can be minimized by setting Motor Deceleration % to exactly 100%.

jussisaily, 2023-05-19:
Thanks for the analysis! I think it's a good explanation, and may be difficult to solve due to the motor being directly connected to the pedals in the coaster brake model. I have the motor deceleration set to 100% already but it had no difference. Like I wrote, someone used the brake sensor for fast stopping. I had to install a front wheel hand brake in this single speed bike (24" Helkama Jopo) for safety, but there is no connection for the brake sensor in my coaster brake motor kit unfortunately. It's not that bad to drive when you get used to it, but being a perfectionist it just keeps bugging me :-) Also I'm sure many people would be happy if there would be some software solution.

emmebrusa, 2023-05-19:
The settings for faster stopping with the coaster brake are those already mentioned. I summarize:
- Use Power assist mode.
- Disable Startup boost.
- Pedal torque adc offset adj = 14.
- Deceleration ramp = 100%.
- Coaster brake torque treshold = 5 (you can try up to zero).

Regarding the brake sensor, if desired it can also be installed with XH18, even without disassembling the controller.
Just use a 6 pin M/F extension between motor and display, locate the two necessary wires (GND and BRAKE) and connect the sensors.

jussisaily, 2023-05-22:
Thanks for the summarized settings! I can confirm that these work quite well! There is no clear forward pedal yank anymore like in torque assist mode, only a slight disengaging delay from the motor. 

Thanks again for your excellent work and support with this FW!