emmebrusa/TSDZ2-Smart-EBike-1 #37: Unable to disable street mode on startup

munterkalmsteiner, 2022-10-06:
Hi,
I'm running v20.1C.2 update 1.
I can compile and flash the firmware (confirmed by changing the number data of displayed at lights on from 3 to 1). However, street mode is always on on startup, even if the option is disabled. I can disable street mode on the display. 

Here's my config.
[config.txt](https://github.com/emmebrusa/TSDZ2-Smart-EBike-1/files/9727721/config.txt)

emmebrusa, 2022-10-09:
Are you sure?
I also tried your config, it works fine.
How do you deduce that Street mode is always active on startup, even when disabled?

munterkalmsteiner, 2022-10-10:
After turning on the bike, the motor stops engaging at 25 km/h. When enabling parameter mode (motor support off, light on/off), and then going to eco mode (light on/off to disable street mode), I get motor support above 25 km/h. I also noticed that enabling "Set parameters on startup" does not work either. I have to enable parameter mode manually as described above. Maybe of relevance: I have a TSDZ2B, i.e. the one with the improved clutch design.

dzid26, 2022-11-05:
Hmm @munterkalmsteiner 
If you happen **_not_** to clear the EEPROM when flashing (like I [do](https://github.com/emmebrusa/TSDZ2-Smart-EBike-1/pull/33)), the ui8_street_mode_enabled flag and some other options will be loaded from EEPROM, and not the config anymore.  So that would match your issue descritpion.

To remember the settings you'd need to write to memory in OFF mode 3rd option:
[image]

munterkalmsteiner, 2023-08-15:
@dzid26, I can confirm that your procedure works, i.e. after enabling streetmode and saving it as default on the screen, streetmode is enabled after a restart. If this is the intended behavior, this issue can be closed. I did not see that in the documentation, though...