emmebrusa/TSDZ2-Smart-EBike-1 #40: First time flashing with VLCD5 display?

kiwisibk, 2022-10-26:
Hello all!

Since I'm getting my sensor extension cable and ST Link USB soon, I wanted to clarify a few questions with you experts.

1. if I want to keep the VLCD5, do I have to do the "conventional" flashing with the STVP software first? Or can I flash directly the first time with the TSDZ2 Parameter Configurator?
2. of course, I was planning to first read out the memory with STVP to see if the connection works, and then save the files from all three tabs in a separate file. Question: would there be a way to reset the engine to the default firmware if these three saved files were lost for some reason? 

Thanks a lot for your help!!! :)

emmebrusa, 2022-10-31:
First save the three files and then you can immediately use the configurator even at the first flash.

kiwisibk, 2022-10-31:
Thanks, I did this, already flashed your excellent firmware, works like a charm!!!! :)