emmebrusa/TSDZ2-Smart-EBike-1 #41: Maximum Values for Assist Modes?

kiwisibk, 2022-10-30:
Hello emmebrusa, first of all thank you for this super software that you provide us!!!!
A question about the values in the respective levels of the various assist modes; if you move the mouse over the respective field comes a note with the maximum permissible values. Are these checked regarding overheating of the motor? Or is it better to set them a little lower to be on the safe side? Thanks a lot!!!

emmebrusa, 2022-10-31:
The maximum values indicated are the maximum accepted by the firmware, they do not have a direct relationship with the maximum power of the motor. The motor can reach maximum power, and overheat, even with lower values.
However, the motor can also deliver maximum power, but for a limited time.

kiwisibk, 2022-10-31:
Understood, I already have an excellent driving experience with levels 2 & 3 set to 150 and 300 respectively in Power Assist, and 100 & 150 in Torque Assist. Turbo level is set to 450 in Power Assist and 250 in Torque Assist. I am using the hybrid assist mode. I think I will only use the turbo stage sporadically on very steep climbs. Do you think I am on the safe side with these settings? And could you show me (if available) a good step by step guide on how to install a temperature sensor and connect it to your software? Thanks a lot!!! :)