emmebrusa/TSDZ2-Smart-EBike-1 #42: "SWIM error 0x05" when flashing "main.ihx" while read&write of original software working

Dizotoff, 2022-11-20:
Hi all, beautiful things you're doing here! 

I followed the OS X and Linux installation guide and got a problem on step 9 while flashing the new software. I tried different commands:

`make -f Makefile_windows flash
`
and
`stm8flash -c stlinkv2 -p "stm8s105?6" -w main.ihx
`
both of them fail with:
```
Due to its file extension (or lack thereof), "main.ihx" is considered as INTEL HEX format!
28634 bytes at 0x8000... SWIM error 0x05
```
Step 8, where I read and wrote the original TSDZ2 software, went well without any errors. I also was able to re-do read&write of the original software after the above-mentioned error occurrence.

This is a pretty strange error for me, could anybody shine some light on what could possibly be the reason for such behavior?

dzid26, 2022-11-20:
I don't know.  I am just gonna say, I was having random errors when the cable was longer than 10cm. Currently I shortened it to 8cm including the plug. 

You can also try out my branch
 https://github.com/emmebrusa/TSDZ2-Smart-EBike-1/pull/33  Someone reported it works fine on Mac

Dizotoff, 2022-11-21:
Hey @dzid26,

Thanks a lot for that link you shared, I found precisely my error code there in the troubleshooting section. The guide suggests connecting `SWIM_RST to the bottom center pin`

I dont have SWIM_RST on my STLinkV2, so I guess I have to order a new one to test it.

closed 2022-11-21 by Dizotoff