emmebrusa/TSDZ2-Smart-EBike-1 #43: No additional power between levels

IArchi, 2022-11-28:
Hello,

I have a TSDZ2 750W, a XH18 display and a 48V 10.5Ah battery.
I've successfully calibrated the torque sensor and got values very closed to the default ones (144 and 297 versus 150 and 300).

I've modified the assistance levels in that way:

| Item         | Power assist mode | Torque assist mode |
|--------------|-----------|------------|
| ECO | 60 | 40  |
| TOUR | 150  | 100 |
| SPORT | 300  | 150 |
| TURBO | 450  | 200 |

Of course, I've enabled *Hybrid assist mode* on startup.

When I switch the motor on using the display button, I have to wait a few seconds before pushing on the pedals. Otherwise the motor does not help. Weird behaviour but I assume this is the way it calibrates. Could you confirm ?

My main issue is when I change assistance levels on the display.
When I start the motor, I'm in level 1 (ECO), pushing on the pedals will provide a certain level of assistance.
**Moving to level 2 or more, does not provide additional power.**
When I climb a small hill for example, I don't have any difference between level 1 and 4, even if I'm only at 17km/h and I push really hard on the pedals. 

I've set street power limit to 750W and battery power max to 650W. Could it be an issue with one of these parameters ?
Do I need to change the battery max current to 18A ?

Thanks for your time.

IArchi, 2022-11-29:
I've found out what the problem is.
Because of merge #33, the EEPROM is not correctly cleared. 

Could you try to revert that or add a flag to reset EEPROM automatically ?

Thanks

emmebrusa, 2022-11-30:
> My main issue is when I change assistance levels on the display. When I start the motor, I'm in level 1 (ECO), pushing on the pedals will provide a certain level of assistance. **Moving to level 2 [...]

The data in eeprom has no connection with the levels of assistance.
If at level 1 you have the same assistance as at level 4, it is because you have already reached the current or power limits set in the configurator (Battery current max, Battery power max or Street power limit).

You should check current and power with an external wattmeter, or on the display.

IArchi, 2022-11-30:
I've made a new test this morning. Levels 1 to 4 do work perfectly but the motor seems to reach the maximum power very quickly.
Isn't the maximum power or max current stored in EEPROM ?

emmebrusa, 2022-11-30:
Battery power max and Street power limit, no.
Instead Battery current max, s is stored in eeprom.
There is no need, I left it as I found it.
Iif you have deleted the eeprom, it is rewritten with the configurator values.
You have to check if it reaches the maximum expected power.
Try temporarily setting:
Data 1 = 4 (motor power)
Time to displayed data 1 = 0 (continuous)
Number of data displayed at light on = 1
By turning on the lights, you will see the power of the motor.
Check the value at various levels.

IArchi, 2022-11-30:
I've flash the motor once again.
Power is really weird.
For example, in level 4 (TURBO), it does not go over 240W.
In level 1, pushing hardly on the pedals, I could not go over 150W.

My ini file:
```
false
true
true
35
false
20
36
297
250
2
30
20
25
15
700
500
13
35
39
100
90
4.35
4.10
4.10
3.85
3.60
3.35
3.94
3.76
3.60
3.44
3.26
3.10
1985
25
false
false
false
true
false
false
false
false
false
false
false
0
false
false
false
false
1
9
10
false
750
25
false
false
47
176
65
85
false
false
false
true
true
false
true
50
false
30
true
false
0
50
50
50
50
50
4
3
10
1
8
0
60
120
200
300
40
80
120
160
80
100
130
160
2
5
8
12
30
35
40
45
60
false
60
15
18
21
24
false
10
144
1
true
false
0
255
true
true
true
false
false
false
20
20
20
34
true
false
false
true
true
false
false
false
```

emmebrusa, 2022-11-30:
In the attached ini file there are several values that are wrong, are they copy/paste errors or are they entered at random?
In order:
You write that you have 48V motor and battery but they are set to 36V.
You have enabled Street mode enable on startup with Street power limit at 250W.
There are unlikely values in the torque calibration parameters and others as well.

If the motor and battery are actually 48V, delete everything and start from the Default_Settings_48V.ini file, then edit:
Wheel circumference = 1985 (verify)
Torque sensor adv. = enabled
Calibrate = enabled
Pedal torque adc offset (no weight) = 144 (verify)
Pedal torque adc max (max weight) = 297 (verify)
Street power limit (W) = 500

Then try...

closed 2022-11-30 by emmebrusa

IArchi, 2022-11-30:
Very weird, if I load the INI file into the configurator, it says 48V.
[image]
I'm going to try that.

emmebrusa, 2022-11-30:
It's true sorry, the copy/paste went wrong for me.
I reviewed your file, you need to enable "Torque sensor adv."
"Battery power max" 700 W is too much for a 504 Wh battery.
I don't see any other problems.

IArchi, 2022-12-01:
I've tried to display torque sensor value during a ride.
It starts at 149 and I can reach 299 pushing on the pedals.
I'm still in hybrid mode and the behavior of the motor seems to be like that:
| Speed | Flat | Hill climbing |
|------|------|------|
| OFF | 0 | 0 |
| 1 | 25km/h | barely 10km/h|
| 2 | 25km/h | barely 15km/h|
| 3 | 25km/h | barely 15km/h|
| 4 | 25km/h | barely 15km/h|

The motor does not provide additional power when I need it.

dzid26, 2022-12-01:
> Speed is not a benchmark for levels. You need to check the power of the motor. Did you try what I suggested?
> 
> > Try temporarily setting:
> > Data 1 = 4 (motor power)
> > Time to displayed data 1 = 0 (continuous)
> > Number of data displayed at light on = 1
> > By turning on the lights, you will see the power of the motor.
> > Check the value at various levels.
> > Besides: Auto dispay data at lights on = enabled
> 
> If you reach max power at level 2, at level 3 and 4 nothing changes. Remember that the maximum power is not the one set! You have set 700W, but to get there you need a high capacity battery with cells with high discharge current. Also the motor would heat up a lot. Consider for a 48V motor, a maximum continuous power of 400W.

Cool tips, I've never tried displaying the power. 
Here is my experience for comparison. 
I have "48V" motor and "52V" battery. It is 14s3p, but it is very Chinese and so I think it has only about 250Wh (15km range).  
Recently I increased the motor power to 600W from 500W because days are cooler now.  On Turbo level (4) the data was showing 59 value. 
The battery is cool to the touch, although I think there is a bit of voltage drop when SOC is low because I think I hit the undervoltage shutdown earlier now.  
I am planning to increase power to 700W once I increase motor cooling, though I would say 600W is quite sufficient already.

dzid26, 2022-12-02:
I played today a bit more with power display.  It is much easier to achieve 600W motor output in torque mode or ebike mode. In power mode it is almost impossible (with default settings). Hybrid is a bit easier than power mode.  
This is simply because human power is the largest at high cadence when motor power tapers off due to BEMF.  Note, _in power mode, motor multiples human power_. 

So as a tip, I would be testing higher motor powers in torque mode, just to be sure it can be reached. Just don't burn your motor. :)

IArchi, 2022-12-06:
The motor easily reaches the max power but when I pedal at around 80rpm, it drops to 60W or around. Even in steep hill, the power is stuck to 150-200W at 80rpm. If I pedal slower, the power increases.
I'm in Hybrid mode but I got the same behaviour in eMTB.

dzid26, 2022-12-06:
> The motor easily reaches the max power but when I pedal at around 80rpm, it drops to 60W or around. Even in steep hill, the power is stuck to 150-200W at 80rpm. If I pedal slower, the power increases. I'm in Hybrid mode but I got the same behaviour in eMTB.

This is normal for tsdz2.  It's due to BEMF. You would need either higher voltage battery (e.g. 52V is slightly better) or lower inductance motor ("36V" motor, but it will produce less torque for the same current). 
There is a lot of posts scattered on endless sphere about people experiences. 

Personally I've got used to it.  And I utilize that as a feature  - when the torque tapers off at high gears I push harder with my legs for few more seconds (increasing RPMs even higher) and then I can do fast, low load shift by quickly reducing legs force. If controller was normally more responsive to torque changes, this wouldn't be needed, but it isn't, so that is a hack for doing fast, low load shifting.  

You can also enable field weakening in Java Configurator, but I disabled it nowadays, because what I described above. And field weakening is not very efficient nor strong.