emmebrusa/TSDZ2-Smart-EBike-1 #47: Please make this project Windows 32 bit compatible - only cygwin folder needs update

heinerolli, 2022-12-10:
Thanks for this great project!!
Everything (compiling and flashing) works also on a 32bit Windows Tablet - exept a tiny part during the file cleanup batch processing. This can be changed by replacing the bin sub-folder contens in the cygwin folder (On my PC this is located here: C:\TSDZ2-Smart-EBike-1-20.1C.2-update-2\tools\cygwin\bin) to the respective 32bis versions.
More details here: 
https://endless-sphere.com/forums/viewtopic.php?f=30&t=110682&p=1741636#p1741636
It would be great, if you could change that so that the project is throughout Windows 32bit compatible.

If you dont want to use the zipped bin file, here are some instrictions on how to install cygwin 32bit:
How to install cygwin 32bit
download setup-x-86 from here:
https://www.cygwin.com/install.html
Install with these options:
C:\tmp\setup-x86.exe --allow-unsupported-windows option --site http://ctm.crouchingtigerhiddenfruitbat.org/pub/cygwin/circa/2022/11/23/063457
(Info from here: https://cygwin.com/pipermail/cygwin-announce/2022-November/010810.html )
Instead of the above adress a local install might work with the local folder http....crouching tiger...
For tsdz2 ofm: Take files which are needed from the ../bin directory and copy them to the project bin directory
(or use bin.zip where the relevant files are in)
[bin.zip](https://github.com/emmebrusa/TSDZ2-Smart-EBike-1/files/10199823/bin.zip)

emmebrusa, 2022-12-11:
Excellent report, I'll try to add it to the next version.

dzid26, 2023-02-02:
https://endless-sphere.com/forums/viewtopic.php?f=30&t=98281&e=1&view=unread#p1746754

We could probably replace /program files (x86) path with `%PROGRAMFILES%`. 

And replace those [cygwin binaries](https://github.com/emmebrusa/TSDZ2-Smart-EBike-1/tree/master/tools/cygwin/bin) with 32bit versions for compatibility.