emmebrusa/TSDZ2-Smart-EBike-1 #48: Error flashing with v20.1C.2-update-2 - stm8s_itc.asm:140: Error: <a> machine specific addressing or addressing mode error

dklausbutt, 2023-01-11:
Hi there @emmebrusa, Thanks for all your great work.
I was trying to flash the new version and during the flash process I get the following error.

sdcc: Calling assembler...
sdcc: sdasstm8 -plosgffwy ""../../bin\stm8s_itc".asm"
../../bin\stm8s_itc.asm:140: Error: <a> machine specific addressing or addressing mode error
removing ../../bin\stm8s_itc.rel
make: *** [../../bin/stm8s_itc.rel] Error 1
Build error!!

I am using the same parameters for my bike as for v20.1C.2-update-1 but simply using the new Java Parameter Configurator 4.3
I even tried to reflash my bike with an older version and it works perfectly. 
Just with the latest version it does not flash my bike.

Is it working for everyone or is it just me getting this error?

Cheers

dzid26, 2023-01-11:
- What OS are you on?
- Do you have the latest SDCC installed?
- Does it work if you remove ` --peep-asm --peep-file peep.txt` from the (makefile)[https://github.com/emmebrusa/TSDZ2-Smart-EBike-1/blob/master/src/controller/Makefile#L69]
- Maybe you can attach `/bin/stm8s_itc.asm` file for us to verify? Mine looks like this [stm8s_itc.zip](https://github.com/emmebrusa/TSDZ2-Smart-EBike-1/files/10395098/stm8s_itc.zip)

dklausbutt, 2023-01-11:
Hi @dzid26,
Answering your questions.
- W10 - 22H2 build 19045.2364
- No, I'm using sdcc-4.0.0 so I will download the latest version and try again.
- I will try if everything else fails.
- my stm8s_itc.asm is different to yours so attaching for you to verify. I will try to flash using your file.
[my_stm8s_itc.zip](https://github.com/emmebrusa/TSDZ2-Smart-EBike-1/files/10395469/my_stm8s_itc.zip)

dzid26, 2023-01-11:
After upgrading SDCC https://sourceforge.net/projects/sdcc/files/latest/download  check if that's your main installation. 

The [script ](https://github.com/emmebrusa/TSDZ2-Smart-EBike-1/blob/master/src/controller/compile.bat#L3)looks onto following paths:
C:\SDCC\
C:\Program Files\SDCC

dklausbutt, 2023-01-16:
Hi @dzid26 that did the trick! Just updating SDCC to the latest version worked like a charm! 
It was my bad for not checking if the other software needed updating. (In this case SDCC)
Cheers!

closed 2023-01-16 by dklausbutt