emmebrusa/TSDZ2-Smart-EBike-1 #49: First time flashing. Question about the stock display.

hideki2, 2023-01-13:
Sorry if it is a very obvious question, but does this mean that I can use stock display. As in dont have to flash the display? I am considering an 850c.

emmebrusa, 2023-01-16:
The OSF Java Configurator version also works with 850C specific for TSDZ2, with 6-pin connector and Tonsheng protocol.
The display does not need to be programmed, only the motor needs to be.
The available features are the same as for other display types.
850C like VLCD6 is awkward in setting the parameters, due to the time of the lights button, but with the correct settings in the configurator it is feasible.

kkcache, 2023-05-03:
I have purchased mine from Eunorau. It is a TSDZ2B motor but I have a BC281 display. The display is set with 5 assist levels from the supplier. Is it possible to use this display with the configurator, as it is not one of the available display options, or do I need to buy an 850C? Or, can I use the configurator to make the settings and just use the display in basic form for turn on, and increase / decrease PAS levels?

emmebrusa, 2023-05-04:
I don't know about the BC281 display, but if it uses the Tonsheng protocol, it should work like all other displays.
Just try. First remember to save the stock firmware.
If error codes are displayed in the speed field, set VLCD5 in the configurator.

kkcache, 2023-05-04:
Thanks. The screen is a 6 pin TFT display that has a setting for protocol; 03 = Tongsheng, 02 = Bafang.

Because I am **really** new to this, here is my understyanding of what I need to do:

1. Download and instal ST Visual Programmer
2. Install the latest Java Runtime
3. Read and save all three data files for the existing firmware; Program Memory, Data Memory and Option Byte. This is my backup of what was there before any changes
4. Run the Java Configurator and use it to make changes to settings.

Questions:

- I'm not sure whether the current settings will be pre-populated or whether I need to enter everything based on the manuals (which I have downloaded).
- Also, do I need to do anything to enable the Java Configurator to be able to Compile and Flash the motor / controller? Do I need to re-flash the controller with a "writeable" version of the firmware?

Thanks for your advice.

kkcache, 2023-07-20:
I have used an 850c display. I loaded everything and it appears to have successfully updated the firmware. However, I have a couple of questions.

 Prior to the firware update, the watts on the display would change as the motor worked harder or less - now it is constantly on 749W. Why is that?

Also, when I turn lights on when on level 3, I get E02 showing which makes me think that its on eMTB. Hiowever, I  chose enable Hybrid on startup. Should it not show E03?

When I try to update the setting to show E03, I turn lights on, then off, then on again - and it flashes E03. I then let it save and turn lights off. If I then check the setting by turning lights on again, it shows E02. What am I doing wrong?

TIA.

emmebrusa, 2023-07-21:
Are you sure the Watts were displayed?
Don't get confused with Bafang?
In the Tonsheng protocol, motor power and current are not foreseen.
As for the riding mode change, (level 2 and 3), E02 E03 E04 indicate the menu item (position), the current riding mode is shown in the speed field.

kkcache, 2023-07-22:
I am sure it is a tsdz2B. I have filmed the motor and the screen and uploaded it to Facebook (https://www.facebook.com/kostas.kalantzis.7/videos/778467797352641/?idorvanity=734929600180813&notif_id=1690008746247447&notif_t=video_processed&ref=notif)

emmebrusa, 2023-07-22:
The video can be seen, but the W values do not stand out well..
Can you confirm that with the stock firmware, the W value changes?

However, the motor power is not foreseen in the Tonsheng protocol.
The only guess is that they have assigned another function to the 2 bytes reserved for VLCD5 to display the TE and TE1 torque values.

Where did you buy the display?
Have you backed up the stock firmware? Can you post the files?

kkcache, 2023-07-22:
I got the display as a replacement from the supplier to the BC281 display because there was an issue with the odometer. They told me the 850c is a more reliable unit. The connector is actually a 5 pin connector. I don't have the expertise on why this is the case but I can defenitely confirm the Watts changes as the motor draws current from the battery. If I am coasting, the wattage is around 150-200W. When I am pushing hard, it can go as high as 750-800W. Normally when I taking off, it is around 450-500W.

[Stock TSDZ2B firmware with 850C display.zip](https://github.com/emmebrusa/TSDZ2-Smart-EBike-1/files/12136504/Stock.TSDZ2B.firmware.with.850C.display.zip)

I took your advice a backed up the stock firmware files. After I tried the Java configurator I restored the original firmware again and it is working well. I'll try to attach them here. Hopefully it will help explain why this is working on the 850C display. I certainly like the feature.

emmebrusa, 2023-07-22:
Okay.
The W values are very high, they don't look real.
Two more questions:
- Is the motor 36V or 48V?
- In the Java configurator, is the display set to VLCD5 or 850C?

kkcache, 2023-07-22:
I have a 500W 48V motor (at least that is what I ordered). The values might be high and wrong because I am not wearing my reading glasses when I am riding - they are defnitely around 750 though at peak.

The files I sent are my stock firmward settings. I used 850C when using the Java configurator (see attached). The wattage dial and number were static at 749W.

[Java ini file.zip](https://github.com/emmebrusa/TSDZ2-Smart-EBike-1/files/12137351/Java.ini.file.zip)

emmebrusa, 2023-07-23:
The stock firmware of that motor has a modified UART protocol.
With the display in hand it would take a few minutes to verify this.
At a distance it is a bit complicated, if you are interested in experimenting we can do some tests.
These are two ebike_app.c files, to be tried one at a time, replacing the standard one, setting 850C in the configurator.

[ebike_app-850C-test-3.zip](https://github.com/emmebrusa/TSDZ2-Smart-EBike-1/files/12138264/ebike_app-850C-test-3.zip)
[ebike_app-850C-test-4.zip](https://github.com/emmebrusa/TSDZ2-Smart-EBike-1/files/12138265/ebike_app-850C-test-4.zip)

You will see static values, I need them to understand which byte is used and if the data sent is the motor current.
If the test gives a logical result, I think it can be modified to display the W.

kkcache, 2023-07-23:
I am happy to help. Is there a way I can interrogate the display and get you the information you need?

With regards to .c files, I am not where they need to go or the steps you want me to follow. I am new to this stuff. Also, when you say static values, where will they be? On the Java app or somewhere else?

emmebrusa, 2023-07-24:
The test to do is simple, here is the procedure.
Open the ebike_app-850C-test-3.zip file, copy the ebike_app.c file contained, in the src/controller folder, you will get a warning that the file already exists and if you want to overwrite it, answer yes.
Then connect st-link to the PC, open the Java configurator and flash the motor, use your configuration (KK Hybrid no street mode + emtb.ini).
Unplug the st-link cable.
Turn on the display, you don't need to test the bike, you just have to look at the W value and take note of it (it should be a value other than 749W and could also be zero).
Turn off the display.
Repeat the same procedure with the ebike_app-850C-test-4.zip file.
I'm interested in the two values of the W, thanks.
With ebike_app-850C-test-3.zip -> xxW
With ebike_app-850C-test-4.zip -> yyW

kkcache, 2023-07-25:
I just did the test. Here are the results:

- 3.zip c file: zero kW shown
- 4.zip c file: 108kW shown

emmebrusa, 2023-07-25:
Well, I hope I have interpreted the test data correctly.
With this new ebike_app.c file, you should display the power in W.
[ebike_app-850C.zip](https://github.com/emmebrusa/TSDZ2-Smart-EBike-1/files/12158351/ebike_app-850C.zip)
The power is overestimated, the displayed value is 108, while the sent one is 100.
The 850C display should also be able to display current (A) instead of power (W).
There should be an item in the setup menu.
If there is, can you check if it works?

kkcache, 2023-07-27:
I used the attached new c file and got the following results (I took photos of the display). I also updated the display to show Watts initially and then Amps.
[image]
[image]
[image]

emmebrusa, 2023-07-27:
30H is a communication error.
I probably did something wrong.
I'll check as soon as I have time.

dzid26, 2023-07-27:
Maybe the error is related to this:
> I also updated the display to show Watts initially and then Amps.

kkcache, 2023-07-28:
I can re-try the Java Config again (I think I saw something about Option byte error when i went into the STCC program). Is the expectation that this latest 850.c file will show both dynamic watts and amps? Should I leave it on and ride with it, or is it more testing to get results?

kkcache, 2023-07-28:
I retried the Java update and it worked fine (must have been a comm's issue given the "Option Byte" error I saw). I went for a small ride and it appears the Watts updates as I ride. I will leave it on Hybrid setting as per the Java config and see how it performs on my next big ride to work on Wed next week. I'll let you know how it goes. Thanks.

emmebrusa, 2023-08-03:
Have you tried showing Watts and Amps?

kkcache, 2023-08-04:
I had only used Watts and it worked well. I just tried amps and it seemed ok, but I just got a error 08H for the first time (not sure what that means) and then no asisstance. It was working well up until then. After a bit, it came back. It did it one more time and then it seemed ok again.

emmebrusa, 2023-08-04:
E08 - SPEED SENSOR ERROR
Faulty speed sensor or magnet too far away, or not centered on sensor (arrow).

kkcache, 2023-08-05:
Thank you. I check the magnet and it had rotated a bit and was not seen by the sensor. I am confused though as I have heard that you can disconnect the speed sensor or remove the magnet on the TSDZ2 and it will work. Is this something that the OSF (Java config) requires?

vasycara, 2025-10-08:
> Am folosit noul fișier c atașat și am obținut următoarele rezultate (am făcut fotografii cu afișajul). De asemenea, am actualizat afișajul pentru a afișa inițial wați și apoi amperi. [image] [image] [image]
> 
> Interesant este că am încercat să văd dacă ar intra în Walk Assist sau în orice altă asistență și nimic - presupun că este de așteptat, nu?
> 
> Între timp, am restaurat firmware-ul original.

hello
I need original firmware for display 850c.
please put here original firmware for display 850c or send me that firmware to maill   vasycara@yahoo.com
Thank you.