emmebrusa/TSDZ2-Smart-EBike-1 #50: Will this firmware work with the TSDZ2B motor

JorgDr, 2023-01-20:
Hi there,

it appears that TongSheng have made a few modifications to their TSDZ2 motor, now calling it TSDZ2B.  The 2 most obvious changes are the clutch removal out of the blue cog and making it solid and white, the other is now installing a double clutch into the main, big cog, thus preserving the overall functionality.

I have no idea if they made changes to the electronics.  Does anybody know?  Will this firmware work with the new motors?

emmebrusa, 2023-01-25:
It also works with TSDZ2B, as long as it's with the old controller v1.
To date, only the motors sold by Enerprof have the v2 controller and therefore not programmable.
There is no other news about it, it is certain that those sold by Pswpower are still programmable.

CableCatDK, 2023-02-20:
I just bought a TSDZ2 from Pswpower, and its firmware is byte identical to all previous models sold by them. So we are still good. But one should consider buying an extra controller, incase production of the v1 controller stops.