emmebrusa/TSDZ2-Smart-EBike-1 #52: it runs hard, then stops for about 5 seconds

chiboreache, 2023-01-27:
is it overheat or bms protection?

[image]

chiboreache, 2023-01-30:
🤔

i'll play with `max A` and `max W` 

noticed just now that i confused battery with motor power

chiboreache, 2023-02-04:
disabled the `street power limitation` and did some changes on the main screen too

but still getting an error and the engine shuts down for a few seconds

it really depends on the load, on a hill with luggage it stalls several times in a row

what could it be? what should i be measuring? btw - i have the DC clamps (A), and there is no speedometer sensor (old firmware works extremely bad with it)

[image]

dzid26, 2023-06-27:
If anything, I would try to reduce battery current. I run it at 8A due to weak battery. 

What error were you getting on the display?

chiboreache, 2023-07-01:
> If anything, I would try to reduce battery current. I run it at 8A due to weak battery.

after getting warm ( above 0-5C ) it runs much longer! so i guess battery is really weak.. 

> What error were you getting on the display?

Err2

dzid26, 2023-07-01:
> Err2

You may be pressing on the pedals during power up

[image]

chiboreache, 2023-07-01:
> You may be pressing on the pedals during power up

always starting at the ground

the only way i can think of - it turns off when i riding hard, and then error popping up, is it viable?

dzid26, 2023-07-01:
If say disconnected torque sensor is high, then yes https://github.com/emmebrusa/TSDZ2-Smart-EBike-1/blob/9f22efbac8baab6c2644afe1cefad0974b1be647/src/controller/ebike_app.c#L1876C36-L1876C36 
But I don't know how it behaves.

Usually it fails partially it seems:
https://endless-sphere.com/sphere/threads/tsdz2-torque-sensor-broken.98665/

chiboreache, 2023-07-01:
so it's T.S fault 🤔

and i probably can't do anything with it, right? since in steep hill or lots of weight, torque would be high no matter what..