emmebrusa/TSDZ2-Smart-EBike-1 #53: v20.1C.2-update-2

69pjc69, 2023-02-08:
Ever since changing to the latest firmware I am occasionally having issues that come and go as they please, I am unable to replicate the issue until it randomly comes again.
The motor will all of a sudden lose all power and display error e04 for a few seconds. I will then stop the bike and the error disappears and I can then continue on my journey. I have the vlcd6 kit with no throttle. I have double checked all my settings and all the throttle options are turned off and it is all set as per my previous setups. Have never had any issues up untill programming this version

emmebrusa, 2023-02-11:
In the latest version, the main change concerns Walk assist, then there are small changes that do not affect the management of the motor.
I think the problem has another origin, probably mechanical. E04 is ERROR MOTOR BLOCKED.
You can still verify this by installing the older version.

69pjc69, 2023-02-13:
Thank you so much for a rapid reply. After hearing from you it is motor blocked error I decided to open up the motor and check for anything broken. Low and behold the bearing in the plastic gear was playing up. Changed it and all is absolutely perfect again. Thank you

closed 2023-02-13 by 69pjc69