emmebrusa/TSDZ2-Smart-EBike-1 #54: Due to its file extension (or lack thereof), "data_empty.ihx" is considered as INTEL HEX format!

Viguro79, 2023-02-12:
Using mac os ventura on macbook pro M1, I'm able to connect the key to the bike.
When lauching ./compile_and_flash_20.sh, it is able to compile ; then ask to backup or not ; then to compile.
```
Do you want to flash the motor ? [Y/n]Y
stm8flash -c stlinkv2 -p stm8s105?6 -w data_empty.ihx
Determine FLASH area
STLink: v2, JTAG: v29, SWIM: v7, VID: 8304, PID: 4837
Due to its file extension (or lack thereof), "data_empty.ihx" is considered as INTEL HEX format!
Address 4000 is out of range at line 1
make: *** [clear_eeprom] Error 255
```
What is this issue related to a data_empty file ?
The TSDZ2-20.1C.2-2-.hex is well generated in the releases folder.

Any idea about this?

emmebrusa, 2023-02-14:
The data_empty.ihx file is correct.
This error message could be a stm8flash bug or the SWIM connection is not working.
I can't verify it because I don't know and don't use MacOS, let's wait for someone more experienced.

PythonNut, 2023-03-12:
This looks like a mistake in the script? I think you need to pass `-s eeprom` to `stm8flash` to write to the EEPROM, which the Makefile suggests is what should be happening.

dzid26, 2023-03-12:
-s eeprom (https://github.com/vdudouyt/stm8flash/blob/7885455458d6b1d351eb039ab88250ce590ab59b/main.c#L361) technically shouldn't matter because data_empty.ihx specifies memory addresses explicitly. 

But maybe it would solve OP's issue. 

Address 0x4000 is indeed eeprom as specified in the data_empty.ihx. 
So perhaps stm8flash is thinking it is programming different cpu?

How about trying with quotation marks? https://github.com/emmebrusa/TSDZ2-Smart-EBike-1/issues/30

Viguro79, 2023-03-26:
None of these combinations work :
``` 
stm8flash -c stlinkv2 -p stm8s105?6 -w data_empty.ihx (original form)
stm8flash -c stlinkv2 -p "stm8s105?6" -w data_empty.ihx
stm8flash -c stlinkv2 -p stm8s105?6 -s -w data_empty.ihx
stm8flash -c stlinkv2 -p "stm8s105?6" -s -w data_empty.ihx
```
Maybe an issue with MacOS

emmebrusa, 2023-03-27:
Try this:
`stm8flash -c stlinkv2 -p stm8s105?6 -s eeprom -w data_empty.ihx`

hrenard, 2023-04-02:
In short, I get :
`stm8flash -c stlinkv2 -p stm8s105?6 -w data_empty.ihx` returns `Address 4000 is out of range at line 1`.
`stm8flash -c stlinkv2 -p stm8s105?6 -s eeprom -w data_empty.ihx` returns `OK`.
`stm8flash -c stlinkv2 -p stm8s105?6 -s eeprom -v data_empty.ihx` returns **`FAIL`**.

I had #37, so I tried to verify the EEPROM was really cleared, and form what I understand, It's not.
For the context, I'm using a STLink-V2 clone in **3.3V** after hitting `SWIM error 0x04`.

Viguro79, 2023-04-18:
Still not working on macos :
```
stm8flash -c stlinkv2 -p stm8s105?6 -s eeprom -w data_empty.ihx
Determine EEPROM area
STLink: v2, JTAG: v29, SWIM: v7, VID: 8304, PID: 4837
Due to its file extension (or lack thereof), "data_empty.ihx" is considered as INTEL HEX format!
```
STLink-V2 clone in 5V or 3.3V, neither works

Ieuanoh, 2023-05-12:
If it's of any use, I was having the same issue on macOS and ubuntu, I applied the fix from [this PR](https://github.com/apolselli/TSDZ2-Smart-EBike-1/commit/ccd959ed845fd8b94b8e3c8844e702b7f3537136) - didn't work, same error. It only started working when I also connected the RST cable - it wasn't super clear for me as [this documentation page](https://github.com/emmebrusa/TSDZ2-Smart-EBike-1/wiki/OS-X-and-Linux-installation) suggests it can be done with just GND, SWIM and 5V. Worked on Ubuntu, didn't get round to retrying on macOS

emmebrusa, 2023-05-12:
Thanks, I agree.
Always connect 4 wires, even RST.

Viguro79, 2023-05-22:
Yes ! I missed this important point : connecting the RST wire did the trick! Thanks everyone

closed 2023-05-22 by Viguro79