emmebrusa/TSDZ2-Smart-EBike-1 #56: Mayor cleanup for throttle control needed

CableCatDK, 2023-02-20:
I have noticed these issues with the throttle control

[1] Safety issue: If ( throttle is not enable street mode ) and ( throttle is enable in offroad mode ) and ( startup mode is street mode ) and ( "set parameters on startup" is enable ): Then the check for bad throttle input is not made at startup. This means that, if throttle is at 100% at startup, there is no error. But if offroad mode is enabled later, the motor runs at 100% suddenly.

[1] Suggested fix: If there is a throttle: Always perform check for bad throttle input. Also perform the check when switching between street and offroad mode.

[2] Issue: If "legal throttle" is enable, it also affects offroad mode. Meaning that the throttle only work if pedaling, even in offroad mode.

[2] Suggested fix: Spilt alle the settings for throttle in a street and offroad mode.

[3] Feature request: Make an option to use the throttle without pedaling, when the speed is equal or below 6 Km/h.

[image]

emmebrusa, 2023-02-22:
Thanks for reporting, anything that helps improve OSF is always welcome.
Something about it, I had already been reported on the Italian forum. I'm already working on it, first with 860C.

> [1] Safety issue: If ( throttle is not enable street mode ) and ( throttle is enable in offroad mode ) and ( startup mode is street mode ) and ( "set parameters on startup" is enable ): Then the [...]
I state that I do not have a throttle installed.
Do you mean the throttle knob could be stuck 100%?
It is an improbable hypothesis but not impossible and it would become dangerous, not only when passing from street mode to offroad mode, but also when the display is switched on, if throttle "without pedaling" in street mode is chosen.
I agree, a thottle input check is needed.
Such a control would also protect against setting errors, if ADC input throttle is chosen when a temperature sensor is installed.

What worries me the most is starting from standstill with throttle in pedaling mode.
There isn't an immediate response that would allow you to adjust the acceleration.
You can bring the throttle knob to 100% without anything happening, starting to pedal, the motor would start at maximum power.
I think proper acceleration management is needed.

Do you have a throttle installed?
How is the start from standstill with throttle in "pedaling" mode?
Can you control the acceleration?

> 
> [2] Issue: If "legal throttle" is enable, it also affects offroad mode. Meaning that the throttle only work if pedaling, even in offroad mode.
> 
> [2] Suggested fix: Spilt alle the settings for throttle in a street and offroad mode.
>
The initial intention was what you suggested, in fact the "Legal" enablement is in the "Street mode" section.
Then I thought about it, having the ability to enable/disable easily from the display would make the accelerator blatantly illegal.
An error was the definition "legal", much better to use something else like you suggested, like "pedaling" and "without pedalig", then the user will have to find out what is legal in his own country and decide accordingly.
I'll fix that too, a little differently than suggested. Instead of offroad I would call it global, because I consider it at a higher level than street, in the sense that if you choose "pedaling" globally (offroad), it will not be possible to set "without pedaling" in street mode.

> 
> [3] Feature request: Make an option to use the throttle without pedaling, when the speed is equal or below 6 Km/h.
> 

From the Italian law which I think is the same as the European one:
Pedal-assisted cycles can be equipped with a button that allows the motor to be activated even when the pedals are stationary, provided that in this mode the vehicle does not exceed 6 km/h.

This is my interpretation:
The legislator does not distinguish between pushing the bike on foot or using it normally on the saddle, it can be deduced that you can circulate without pedaling up to 6 km/h.
However, technically, an accelerator is not a button.
A push button is a device with two positions, one of rest to which it returns when released, and one when it is held down.
With a push of a button, acceleration and motor power are predefined by the manufacturer.
With an accelerator it is the user who has to manage them, they could be too high and dangerous.
Perhaps my interpretation is too rigid, I don't think of adding this feature, at least until starting from a standstill with 100% accelerator will not be safe.
I'm going to make a change to the already existing "Startup assist" feature instead.
To avoid it being used as a throttle, I had set a time limit of 10 seconds, now I will remove the time limit and add a speed limit of 6 km/h.

A curiosity, I have not grasped the meaning of the title.

CableCatDK, 2023-02-22:
First: Thank you how a well written response, and for spending your free time on this project.
I will answer the different issues in individual posts here.

CableCatDK, 2023-02-22:
About "throttle knob stuck at 100%", I can think it happing like:
- Mechanically/physical stuck: This can happen simply by jamming components too tight on the steering wheel. It is easy to see and correct however.
- Electrical short: Poor electrical work. Bad component. Water intake. See my story below:

Last spring I installed a TSDZ2 + VLCD5, where I shorten the cable between them. I did not install a throttle, and did not intent to. I had not yet install the ebreaks, because I was testing it.

I told the firmware I had ebreaks and a throttle, because I was using previous settings. Of cause I should have removed the throttle from the firmware, which I did later, after the intendent describe below.

A intendent happen, where the bike had been standing out in the rain. After which the none existing throttle got stuck at 100%. I don't exactly know why. When I turned on the bike, everything was normal. When I switched to offroad mode, the motor started at 100%.

Fortunely no accident happen. Because my mechanical break where strong enough to stop the bike, and the power switch to the battery was easily accessible.

The shorting of the cable can be seen here (in Danish). I by mistake also disassembled the display.
https://youtu.be/a_EeKsCCi74?t=398

CableCatDK, 2023-02-22:
About "The legality og using throttle when speed it 6 km/h or below", I can tell how the EU law is implemented in Denmark:

Translation of [Bekendtgørelse om cyklers indretning og udstyr m.v., chapter 9](https://www.retsinformation.dk/eli/lta/2016/976#K9):
> § 27. A bicycle may be equipped with an electric auxiliary motor and may not be equipped with other motors.
> 
> PCS. 2. An electric auxiliary motor must have a maximum output of 250 W.
> 
> PCS. 3. An electric auxiliary motor may only produce power at speeds of 25 km/h or less.
> 
> PCS. 4. An electric auxiliary motor may only produce power when the bicycle's pedals or the like are simultaneously operated.
> 
> PCS. 5. An electric bicycle can, however, be equipped with an electric walking function, so that the bicycle's electric auxiliary motor can produce power at speeds of 6 km/h or less, even if pedals or the like are not simultaneously operated, when an electric switch, f .eg push button or **rotary handle mounted on the handlebar**, is manually affected by the driver. The power output of the electric auxiliary motor shall automatically cease when the influence of the electrical contact on the handlebar ceases.

So the Danish implementation gives a "rotary handle mounted on the handlebar" as an example of allowed switches to turn on walk assist.

CableCatDK, 2023-02-22:
My use case is that I would like to legally use throttle on my Nihola cargo bikes to:
- Fast start from standstill. Very useful when starting from a traffic light.
- Use max assist, when going steep uphill, or ghost pedaling when tied.

Right now I am using version "34 - 2021/10/25 - Revert v20.1.C.1" on my bikes. I am using throttle, without the need to pedal. I have changed the "Motor acceleration (%)" from the default 35 to 28. This gives a smoother start, and is less stressful for the transmission. I don't always use 100% throttle at startup. If I need a very fast start, I engage the ebreak, then the throttle at 100%, then I press the foot down, and disengage the ebraek at the same time.

I have not tried "throttle when pedaling" yet. But I am making a new bike, and will try the latest firmware soon.

I think you give the user the freedom to configure the firmware how they like. But provides good defaults, that are legal in the most of the EU.

dzid26, 2023-05-19:
> I'm going to make a change to the already existing "Startup assist" feature instead.
To avoid it being used as a throttle, I had set a time limit of 10 seconds, now I will remove the time limit and add a speed limit of 6 km/h.

[image]
`#define STARTUP_ASSIST_ENABLED 1`

For me Startup assist doesn't work. - Instead walk assist turns on even with light on.  I looked in the manual and in the code and it shouldn't be like that so I am not sure what is going on. 
Anyway, I don't care about this feature currently, just reporting as a possible issue.

amisi97, 2023-05-27:
I have had an issue.

I use the throttle, sometimes, i do know really when, i stop using the throttle, the bike keeps giving power.

I set

 I can turn off the battery and controller, when I turn on again. I will have the same issue. So this looks like an hardware issue.
  When I unplug the throttle, it finally stops. However, I have used my throttle on the original firmware without any issues so something makes me also think it does not come from the throttle itself.

Do you know what is happening ?

dzid26, 2023-05-27:
@amisi97  doesn't this issue (and yourself) already suggest a mechanical problem and that software checks are not in place? 

> About "throttle knob stuck at 100%", I can think it happing like:
> Mechanically/physical stuck: This can happen simply by jamming components too tight on the steering wheel. It is easy to see and correct however.
> Electrical short: Poor electrical work. Bad component

And
> I agree, a thottle input check is needed.

amisi97, 2023-05-27:
Might be a Bad component.

For the mechanichal issue, i am aware about what is the mechnical problem with this throttle and i know this is not the issue (it can happen the throttle is stuck in down position when the screw is too tight  but this is not the case here). I will try to test with a multimeter when I see i have the problem again. Let me know if you have had a similar issue with this TSDz2b throttle.

emmebrusa, 2023-05-28:
> I have had an issue.
> 
> I use the throttle, sometimes, i do know really when, i stop using the throttle, the bike keeps giving power. It is very random but happens quite often.
> 
> I set ADC throttle value at 120 - 200 and assist throttle value at 0 - 255
> 
> I can turn off the battery and controller, when I turn on again. I will have the same issue, directly the motor starts to give power without any action. So this looks like an hardware issue. It is NOT a mechnical problem on the throttle Only way to stop,was to unplug the throttle. However, I have used my throttle on the original firmware without any issues so something makes me also think it does not come from the throttle itself.
> 
> Do you know what is happening ? Difficut to tell because it happens so randomly.

Why 120 - 200? Is this data that you have detected?
Try this, in the configurator set provisionally:
Data 1 = 5 (adc throttle)
Time to displayed data 1 = 0 (no time limit)
Number of data displayed at lights on = 1

Then turn on the display and turn on the lights. Ignore the decimal point.
With throttle at zero, the displayed value is to be entered in "ADC throttle value min".
With full throttle, the displayed value is to be entered in "ADC throttle value max".

Throttle input control at power on will be in the next version.

emmebrusa, 2023-05-28:
> [image] `#define STARTUP_ASSIST_ENABLED 1`
>
> For me Startup assist doesn't work. - Instead walk assist turns on even with light on. I looked in the manual and in the code and it shouldn't be like [...]

I checked, indeed there is a problem.
I didn't notice because I use 860C.
Thanks for the report.

amisi97, 2023-05-28:
> > I have had an issue.
> > I use the throttle, sometimes, i do know really when, i stop using the throttle, the bike keeps giving power. It is very random but happens quite often.
> > I set ADC throttle value at 120 - 200 and assist throttle value at 0 - 255
> > I can turn off the battery and controller, when I turn on again. I will have the same issue, directly the motor starts to give power without any action. So this looks like an hardware issue. It is NOT a mechnical problem on the throttle Only way to stop,was to unplug the throttle. However, I have used my throttle on the original firmware without any issues so something makes me also think it does not come from the throttle itself.
> > Do you know what is happening ? Difficut to tell because it happens so randomly.
> 
> Why 120 - 200? Is this data that you have detected? Try this, in the configurator set provisionally: Data 1 = 5 (adc throttle) Time to displayed data 1 = 0 (no time limit) Number of data displayed at lights on = 1 Set parameter on startup = disabled Auto display data at light on = enabled
> 
> Then turn on the display and turn on the lights. Set to 0-OFF level. Ignore the decimal point. With throttle at zero, the displayed value is to be entered in "ADC throttle value min". With full throttle, the displayed value is to be entered in "ADC throttle value max".
> 
> Throttle input control at power on will be in the next version.

I put 120 - 200 because when i had the problem with the throttle, i was thinking first the controller did not detect that my throttle was up again. Also, I think i prefer having assistance once I really pushed on it and not if I give a little hit not voluntary. Finally, because it is a bad quality component (the worst of the kit provided by pswpower), it sometimes stays in the same position or not totally up so if it 's between 0 and 120, I will not have problem.

I will try to do what you told me to study the throttle. I come back to you once I have done. I did not have the issue again today but twice yesterday.