emmebrusa/TSDZ2-Smart-EBike-1 #57: Battery %SOC on TSDZ2B with VLCD6 display

pickerdave, 2023-03-08:
Hi - my first comment, please be gentle!!! :)
I have a TSDZ2B 36V 350W with VLCD6 display. My configuration is as the attached.
[image]
The battery %soc displayed at start-up and when the lights button is selected always reads 99.9. There is some fluctuation but it settles on 99.9. The battery voltage reads steady at 41.4V.
Is my set up correct or is there a better way?
Does the soc feature operate corrected or is there a known issue?
Thanks

dzid26, 2023-03-08:
Assuming you have a "36V battery" (10cells in series), then 41.4V/10 is 4.1V. Thats fully charged. 100% cannot be displayed, so it is showing 99.9.

pickerdave, 2023-03-08:
Thanks for your reply.
I've used the table to correlatevoltage and soc and it does seem to be correct. 
The soc indication does jump around a little but is certainly good enough.
Thanks

[image]

closed 2023-03-10 by pickerdave