emmebrusa/TSDZ2-Smart-EBike-1 #59: possibility to change Amps from vlcd5 (if not)

aston89, 2023-04-09:
hi mbrusa, im wondering why i cant change the amps drawn directly from the display ? it's a feature already present in your firmware i tought

dzid26, 2023-05-02:
You could just change assist level, no?

aston89, 2023-05-12:
Just wanted to limit the amps witouth having to reflash or navigate trought the settings panel to switch profile (it's a bit over complicated when you dont do it regularly).

In the standard setting panel of vlcd5 there is this possibility but with this firmware it doesnt affect the amps anymore.

I am asking this because mbrusa was able to link the torque check parameter integrated in the display in the recent changes so i tought it was possible to link this too.

Obviously i have a suspect that this is not easily doable due to the fact that the firmware configurator does not ask you to configure the Amp but to configure the overall Wattage, it does take in consideration volt and amp togheter due to the extended functionality of the motor (variable voltage, field weakening and more) so maybe, assigning a specified wattage value for every amp in the display could be a solution to manage it (with a table of conversion / comparison).

dzid26, 2023-05-12:
> due to the fact that the firmware configurator does not ask you to configure the Amp
[image]

It doesn't ?

> so maybe, assigning a specified wattage value for every amp in the display could be a solution to manage it (with a table of conversion / comparison).

What are you trying to achieve in general?

aston89, 2023-05-13:
That parameter exist but the documentation (and the suggestion bubble) ask you to directly put a specific amount of amps for 36v (17a) or 48v (12a).
Under that parameter you will find the one involved in the wattage, in this case the description is saying that it will be the maximum power drawn by the combination of the motor and the possibility of the battery.
I logically assume that the parameter involved in the maximum current absorption (volt x amps) is the watt one but in the display (as i said before) you can only change on the fly the amp parameter.
That parameter seems to not be linked anymore to any setting in the firmware, so, i was asking and thinking if the reason could be that the mbrusa-firmware is reading a watt-value instead of an amp-value internally.

But honestly at this point i have probably created more confusion than before, i just want to be able to change the amps/watt while riding using the classic method of configuring the vlcd5 instead of digging into the more complicated multi layered settings that i have to memorize and forget every time i need to use them

dzid26, 2023-05-13:
1. First of all, I don't think [VLCD5](https://wooshbikes.co.uk/manuals/VLCD5.pdf) [has](https://custombuilt-ebikes.co.uk/wp-content/uploads/2020/09/VLCD5-display-manual-for-TSDZ2.pdf) current setting. 
2. You CAN change the battery current setting (which for the constant battery voltage is linearly related to motor current) which will influence maximum torque (which for this motor happens at a relatively low cadence). Though I don't see a reason to set it to anything else but a maximum of what your battery can provide. 
3. Just use assist levels to "change" the power on the fly. The power will be proportional to assist level x pedal force, though exact values. You can also switch between maximum power levels by configuring `Battery power max` and `Street power limit` and then switching between street mode and offroad on the fly. 

> Under that parameter you will find the one involved in the wattage, in this case the description is saying that it will be the maximum power drawn by the combination of the motor and the possibility of the battery.
It says that this value should be chosen based on motor-rated power (too not overheat it too fast) and battery capacity (to not discharge it too fast). 

Note: 
battery power = motor power;  
battery current ≠ motor current (but they are proportional to each other for constant battery voltage).

In summary, set the battery current to what battery can do (if it shuts down under load (due to undervoltage) that means the value is way too high). Then set the `battery power max` and `street power limit` according to your preference (e.g. 500W and 250W).   

You can observe actual power in the display by doing [this](https://github.com/emmebrusa/TSDZ2-Smart-EBike-1/issues/43#issuecomment-1333792432) in advanced configuration. 
>`Data 1` = 4 (motor power)
`Time to displayed data 1` = 0 (continuous)
`Number of data displayed at light on` = 1
By turning on the lights, you will see the power of the motor.

aston89, 2023-05-13:
1 vlcd5 have this setting under the secret menu that you can access holding the two button togheter, it also have another menu for reading the initial torque value and current torque reading from the sensor

2 ok so my battery can provide a 1000w continuous, its supposed to be something like 20a or 22a but the fuse say (if i remember correctly) 36A, i guess il have to augment it ?  seems odd 

3 it's difficult to explain why in certain situation i dont need all that power but only a fraction at low pedal torque but ok anyway, as i said the mrbrsa settings are overcomplicated for me to use and to remember, the secret menu on the display is brilliant tricky but also very difficult to manage (in fact i disabled it) 

by the way, thanks for the explanation about the correlation about the battery amp and the voltage now i understood that parameter better than before

dzid26, 2023-05-13:
1. Reference?
2. You can set 20A. It will be able to produce a lot of torque at a low cadence. At higher cadence current tapers of due to BEMF
3. Assist level 1 should do the job. You can lower its setting at `Assistance Setting` tab in java configurator. What mode are you on? I hope not Cadence mode...

aston89, 2023-05-13:
1: here you go https://www.youtube.com/watch?v=7ycRYVKRBtk  (various parameter)
there is also another menu similar to this where you can read initial torque value and realtime, in one of the latest updade you will read in the changelog that mbrusa finally fixed or linked that function, before that update it was disabled so i assumed that there is the possibility to link also the amp parameter in a future update maybe.

2: ok but isnt supposed to be the maximum allowable for the controller also ? when the battery is full charged it does have 54v, multiplied by 20A is something like 1000w, as far as i know that current will have to pass somewhere, so it's limited internally by the watt parameter ? sorry for redundantly asking the same questions in a different flavour but i have to be sure to not fry anything

3. i have the "power assist mode" 

now im having a doubt, does the "torque assist mode" reads the amp value and the "power assist mode" read the watt value from the basic setting panel ?

dzid26, 2023-05-14:
1. https://www.pedelecs.co.uk/forum/threads/tsdz2-what-does-assist-ratio-power-setting-in-vlcd5-hidden-menu-change.34269/
2. The MOSFET is rated 80A, so it's rather ok to pass 20A. That is going to be limited by `battery power max` setting ot simply by level of the assist. 
3. I don't actually know if assist levels are related to the `battery power max` (or other setting) or whether they are absolute values. I was wondering that myself because my 3rd (hybrid) assist level is very similar to 4th, so I am thinking the levels are somewhat absolute. Would be good to crosscheck the manual with code.

aston89, 2023-05-14:
1: i am surprised, due to bad documentation and tutorial videos i always assumed that this parameter on the display changed the Amps, that's explain why i tought it was possible to modify the power throught it, got it now.

2: ok that's good i didnt knew or read anywhere that the mosftet was capable of 80A (i did fried two controllers due to mechanical failures that lead to sudden motor stop during motion so that's why i am being kinda paranoid) 

3: well the main "problem" was basically solved so i think well figure out in the future, thanks for the info

closed 2023-05-14 by aston89

dzid26, 2023-05-18:
> * The MOSFET is rated 80A, so it's rather ok to pass 20A. That is going to be limited by `battery power max` setting anyway or simply by level of the assist. And you probably would only see 20A flowing at very low speed and not in power assist mode.

Just a side note - you may not want 20A limit for two reasons:
1. The blue gear may not handle 20A worth of torque
2. The torque response will be very cadence dependent (red instead of blue):
[[image]](https://www.motioncontroltips.com/when-is-motor-output-power-useful-specification-how-is-it-calculated/)

I recently ride with only 8A torque - partially because my small battery is getting old and weak and partially because I am enjoying a flatter torque curve.