emmebrusa/TSDZ2-Smart-EBike-1 #64: E02 E03 E04 are not flashing

amisi97, 2023-05-26:
Hello, i have VLCD5 display.
I just download master branch code and give the firmware into my motor.
According to the manual, when i am at off level, if i press on/off, i should see E02 flashing. What I see is E02, and i if i press on/off again, i can go to E03 and then E04. But none of them are flashing. How can i make it flash so it  E02 can be enabled and then i can access other mode.

Regards

amisi97

amisi97, 2023-05-27:
I enabled "Set parameters on startup" on the configurator. It is working good now. I see now flashing 5 sec and then it is validated.

amisi97, 2023-05-27:
More info about the issue. I can always make E0X appear. But sometimes, it is blocked, you cannot make the change. You have restart controller again and again and it will be unblocked at one moment after several restart or minutes.

dzid26, 2023-05-27:
> More info about the issue. I can always make E0X appear. But sometimes, it is blocked, you cannot make the change. You have restart controller again and again and it will be unblocked at one moment after several restart or minutes.

[image]

I can enable this and then disable it without restarting the controller.

Something tells me you just need to learn more how to use those error displays - they are not intuitive at first.

amisi97, 2023-05-27:
You might be right, let's wait some time for me to learn. I will see if I still have some issues after that learning phase.

amisi97, 2023-05-31:
After some time using the controller, I still do not understand how to control it properly.

Let's say, i am on level 2. So I could chose between E02(power assist), E03 () , E04 ().I can make them appear on the by double tapping the light button. On this, I never have issues. However,when I stop let's say on E03, I wait, and no flash of E03 appear. Do you have to validate or something ? 

When there is the E01,2,3 flashing, i know it has changed and do not fail. My problem is I select the mode but it is finally not flashing.

Same problem at other level, inclued OFF level.  Once I selected the mode, if do not press anything ,it should flash and then be selected. But this does not happen ! Sometimes it worksafter repeated tries so I can make my change but most of the time, no.

I might do something wrong.

I took a video if necessary.

Regards

amisi97

amisi97, 2023-05-31:
I can confirm it is working good now. However, it is very non user friendly for beginners. 

So for beginner with VLCD5,here are some advice you should know:

You have 2 kind of settings , BINARY and OPTION.

- LEVEL 0 and LEVEL OFF mode are BINARY. Each mode on this level has to be enabled or disabled. Changing a binary status won't affect any other binary status.
- LEVEL 2  and LEVEL 3 give you 6 options. You can only be at once option a time among those 6.
- LEVEL 4 gives 3 options. You can onlybe at once option a time among those 3.

LEARN ABOUT A PARAMETER :

- You have to make sure you currently are in SET PARAMETER ENABLED. Turn on the light, if E0X appear at some point, it means you are good.
- For an option, just turn on the light, it will show the current parameter used E01, E02 or E03. However, you cannot know if it is the one from level 2 or level 3.
- For a binary parameter, select the level 0 or OFF according to which parameter you want to know, and then select this parameter by double tapping the light button. Once you are there, wait a few seconds and DO NOT light off the screen, E0X disappears, then you can see either "10.0", it means enabled. If then, you see "00.0", it means disabled.

CHANGE A PARAMETER  : 

- You have to make sure you currently are in SET PARAMETER ENABLED. Turn on the light, if E0X appear at some point, it means you are good.
- Chose the level according to whatever parameter you want to set and then select the right E0X parameter by double tapping the light button. Once you are on the chosen parameter. Validating is done by turning OFF the light so turn off the light and wait for at least 5 seconds. E0X should flash 2 or 3 times on the screen. At the end of flashing, it means you made the change.

You can verify it by following LEARN ABOUT A PARAMETER instructions.

dzid26, 2023-05-31:
> For a BINARY parameter, select the level 0 or OFF according to which parameter you want to know, and then select this parameter by double tapping the light button. Once you are there, wait a few seconds and DO NOT light off the screen, E0X disappears, then you can see either "10.0", it means the selected binary setting is enabled. However, you see "00.0", it means disabled.

You problem used right, but it should be written that tapping odd number of times will display the parameter state. If you will "double" tap the parameter will be toggled. 

And of course, I like to print out this page: 
[image]