emmebrusa/TSDZ2-Smart-EBike-1 #65: Set maximum speed from display - dashes only?

kiwisibk, 2023-05-28:
I recently reflashed my wife's and my TSDZ2 and selected the option to take the speed from the display. Everything works fine on my wife's bike, but on my VLCD5 display there are only four dashes when I want to set the speed. Who can help me?

kiwisibk, 2023-05-31:
It's crazy, I flashed the original files again with ST Link, no problems. But in the VLCD5 display still no speed can be set, only four dashes are there. Therefore the engine does not run at all with the original formware, because it does not recognize any speed. Then I flashed Emmebrusa again and set the speed in the configurator, then everything runs again without problems. It must be the display.

dzid26, 2023-06-01:
@kiwisibk are you talking about this or what?:
[image]

kiwisibk, 2023-06-01:
I mean this: [image]