emmebrusa/TSDZ2-Smart-EBike-1 #68: VLCD5 + Modes

mcnelson, 2023-07-05:
Hi, I am trying to use a VLCD5 screen with the custom firmware and I'm having a hard time finding out how the many modes in the configurator Java tool work with it. It seems this software is optimized for the 860C screen. I have flashed successfully using a VLCD5 screen, but I'm not clear how it behaves with no ability to switch modes.

Does just one of the modes (power/torque/cadence etc) work with the VLCD5 screen? If so, which one? Just trying to get a little more power out of the motor and not sure how to configure it.

dzid26, 2023-07-05:
All of them work. You should read the [manual](https://github.com/emmebrusa/TSDZ2-Smart-EBike-1/blob/master/manuals/EN-Operating_manual_of%20_display-TSDZ2-v20.1C.2.pdf).

mcnelson, 2023-07-05:
Ah, this is what I was looking for! Thanks!

closed 2023-07-05 by mcnelson