emmebrusa/TSDZ2-Smart-EBike-1 #69: Question about the Battery capacity (Wh) input field in the Parameter Configurator

kiwisibk, 2023-07-06:
Hello,

what should I enter in the battery capacity (Wh) input field in the parameter configurator if I have a 19.2Ah 48V battery?

Exactly the value 19.2 x 48 = 921.6 Wh?

The comma as a point or semicolon?

Or should I round the value to 921 or 922 Wh?

How accurate does this input need to be?

Thanks for the explanation!

dzid26, 2023-07-06:
Use integers. 
SOC calculation is not accurate.

kiwisibk, 2023-07-06:
Ok thank you!

emmebrusa, 2023-07-07:
The nominal capacity of the battery (Volts x Ah) must be entered.
Must be an integer with no decimals, round as you like, it doesn't matter.
It is used, together with "Battery capacity calibration (%)", only to calculate the % remaining capacity of the battery.
Read the manual carefully about it.

kiwisibk, 2023-07-07:
Ok thank you, I entered now 920 Wh rounded. (48V x 19.2Ah = 921.6 Wh)