emmebrusa/TSDZ2-Smart-EBike-1 #70: Which 860C Display Version I should Install?

RichardIstSauer, 2023-07-07:
Hello I installed the newest firmware to the motor v20.1C.2-update-2. But which firmware I should install to the 860c display? Can I take also the newest, v20.1C.4-860C-update-2?

emmebrusa, 2023-07-10:
This v20.1C.2-update-2 version is only for stock displays,
not good for 860C.
The most recent for 860C is this:
[https://github.com/emmebrusa/TSDZ2-Smart-EBike-860C/releases/tag/v20.1C.4-860C-update-3](url)

RichardIstSauer, 2023-07-10:
Okay so I can use the latest stable v20.1C.4-860C-update-2? But how I should flash the motor there is no Java Configurator?

emmebrusa, 2023-07-11:
Update 3 is also stable, we've been testing it since April.
Java configurator should not be used with 860C display.
To flash the motor, use the .hex file and ST Visual Programer with ST-link.
To flash the display, use the .bin file and Apt Burn Tools with bootloader box.

RichardIstSauer, 2023-07-11:
Is there a guide for flashing the .hex to the motor? And how I can change the settings when using the hex file, like activating the hybrid mode or other things?

RichardIstSauer, 2023-07-29:
Flashed without any problems

closed 2023-07-29 by RichardIstSauer