emmebrusa/TSDZ2-Smart-EBike-1 #72: Some errors during compiling procedure

Barell69, 2023-08-01:
Hello guys, I get some ebike_app errors during flashing, can me someone help with this potenciall issue ? The flashing procedure afterwards runs ok, but I am not sure, if all parameters are flashed properly. many thanks

[image]

dzid26, 2023-08-04:
Confirmed with 4.3.2. 

Adding `void` in empty function arguments removes the warning. https://stackoverflow.com/a/47693
[image]

Barell69, 2023-08-04:
okey :) 

In which file should I make this change - ebike_app.c ?
After overwriting all static void's still two messages alive :)

[image]

Barell69, 2023-08-04:
I have overwritten also this two lines with "void" and now I am without all errors

uint16_t read_battery_soc(void);
uint16_t read_battery_soc(void)

I hope all these changes don't have any impact to the programming process!
Many thanks  guys !

chgpalmer, 2023-08-24:
It won't make any difference to the resulting binary in this case, but yes in C this:
```void foo();```
Means foo can take any number of arguments

If you have a patch for this already I suggest creating a PR