emmebrusa/TSDZ2-Smart-EBike-1 #78: Issues compiling / flashing

dduhduh, 2023-09-05:
I am getting this error when trying to flash. Any suggestions?

[image]

dduhduh, 2023-09-05:
I also tried using previous release, and while it was able to "clean" it then had a different error. 

MAKE Version 5.41  Copyright (c) 1987, 2014 Embarcadero Technologies, Inc.
"Cleaning files..."
"Done."
MAKE Version 5.41  Copyright (c) 1987, 2014 Embarcadero Technologies, Inc.
Fatal: '../common/STM8S_StdPeriph_Lib/src/stm8s_iwdg.rel' does not exist - don't know how to make it
Press any key to continue . . .

So I'm stuck now...

kiwisibk, 2023-09-06:
Try to reinstall SDCC.

dduhduh, 2023-09-06:
> Try to reinstall SDCC.

I tried reinstalling SDCC. I am installing it directly to C:/ I have tried for both 4.3.0 version of SDCC and 4.1.0 . 

_What command are you running?_ - From the paramater configurator when I select "compile & Flash" I also get the same result when running compile_and_flash_20.bat
_What OS?_ - Windows 10 OS Build 19045.3324 Using a 64-bit 
_What SDCC version?_ - I have tried for both 4.3.0 version of SDCC and 4.1.0 both were installed directly to C:/ when tried. 

Also github zip file is unzipped directly to C:/ 
STMircroelectronics installed directly to C:/

kiwisibk, 2023-09-06:
AFAIK SDCC doesn't need to be installed directly to C:\.. just the ST Visual Development need to be installed directly to C:\. Read carefully the wiki.

dduhduh, 2023-09-06:
> 

I've tried uninstalling and reinstalling SDCC into the default install location C:\Program Files\SDCC 
Still facing the same issue...

kiwisibk, 2023-09-06:
Did you try to reinstall all three apps SDCC, ST Visual Developement and Emmebrusa Software?

dduhduh, 2023-09-06:
Yes . I completely deleted all existance of the Emmebrusa software folder downloaded from Github from downloads C drive and cleared recycle bin. Reinstalled directly to C:/ and still does not work. With the latest version of Emmebrusa software I have the following error. 
MAKE Version 5.41  Copyright (c) 1987, 2014 Embarcadero Technologies, Inc.
Error makefile 92: Too many rules for target '../../bin/%.rel'
Error makefile 97: Too many rules for target '../../bin/%.rel'
*** 2 errors during make ***
Cleaning failed!!
Press any key to continue . . .

With the previous version It completed cleaning but had issue with converting .c files to .rel files which sounds like an SDCC issue so I uninstalled/reinstalled with two different versions and still receiving the same errors.

dzid26, 2023-09-06:
Somehow your Embarcadero Make takes over. I would use GNU make that is included in tools/cygwin.

Also, use proven SDCC 4.2. The latest one has a [problem](https://github.com/emmebrusa/TSDZ2-Smart-EBike-1/issues/76).

dduhduh, 2023-09-06:
@dzid26  Would I just double click  make.exe in tools/cygwin ? I do not know how to not make Embarcadero Make run? 
I will try using SDCC 4.2

dzid26, 2023-09-06:
> @dzid26 Would I just double click make.exe in tools/cygwin ? 

No.
Maybe you can try to uninstall the other make.

dduhduh, 2023-09-07:
Okay so update! I followed exact same steps on a different computer and it worked! At least it compiles.... now I'm faced with a new issue... 
[image]

I am using short ~5cm long jumper cables for making connection to speed sensor. I am using rst , swim, ground , 5v pin . 

And I am able to use STVP to read firmware on motor. However this only works after hitting read all 3 or 4 times. 
When flashing from the java program when I start flash the ST-Link v2 blinks once and then I get the error message. 
When reading from STVP when it finally reads, it blinks many times.

dzid26, 2023-09-07:
I use 5v as well and I find it 100% reliable even with 30cm cable. But I have to make sure the bike charger is disconnected.
Maybe display should be off as well.

dduhduh, 2023-09-09:
I figured out the issue to the second issue I was having. I was doing everything right except the cheapo st-link v2 was no good even though it was able to read after 3 or 4 tries in STVP it would never work first try so it would fail when trying to flash not exactly sure what was wrong with it.... I ended up purchasing an alternative st-link v2 mini that looked different and was just a lot better quality. It worked with the same USB drives and everything. First try worked perfect. I purchased it from Amazon here is the link Mini ST-LINK/V2 SWIM / SWD in-circuit debugger/programmer for STM8 and STM32 @XYGStudy https://a.co/d/5kgc9qi

The manufacturer mentioned on Amazon is not actually the manufacturer it is rebranded.. link to actual manufacturer website.. https://www.waveshare.com/st-link-v2-mini-stm32.htm

Loving the new firmware

closed 2023-09-11 by dduhduh