emmebrusa/TSDZ2-Smart-EBike-1 #83: Make a bafang-like cruise function?

mishuha, 2023-09-24:
I've tried to use the cruise mode as is and I found it very disappointing. It's required long pressing of a button (the same can be reached by pushing throttle) or idle pedaling.  Riding tens of kilometers holding something pressed is exhausting for hands. I'd like to avoid this.It also leads to losing of dosed pedal-assisting benefits when it is required because of that cruise is an separate mode excluding assist. 
I have not found also how to regulate the power increment stepping and motor turning on to reach the speed feels like an impact every time. 

With the bafang mid-drive a cruise mode is implemented as a function, not like an additional mode. That is much more convenient.

So the script is:
1. Get the required speed by pedaling or by throttle bar;
2. Press cruise button to fix the speed (could be holding a "walk assist" button on TSDZ);
3. Stop pedaling or release the throttle bar;
4. Now the bike in the cruise mode.

To switch off: touch throttle again or touch brakes or press cruise button. And it is available with any assist mode. 

And 1st point as alternative could be realized just like now: with level-depending predefined values.

Is it possible to implement on TSDZ?

emmebrusa, 2023-10-20:
I'm answering now because I was on break this summer.
I'm surprised there is anyone who uses cruise, even more so with throttle installed.
The logic you described was present in the versions previous to mine (in the 860C version it is still like this), but always holding down the walk assist button.

In the version for standard displays, I preferred to make a separate mode, with more features.
- minimum cruise activation speed = 0
- possibility of maintaining speed without holding down the button, but only by moving the pedals.
- possibility of changing speed while driving with + and - keys

I don't know what Bafang is like, but I think maintaining speed without holding down a button or throttle is very dangerous.
Please note that TSDZ2 installations are almost all brake sensorless.

Maintaining speed without holding down a button or moving the pedals is against my safety principles.

What isn't good in cruise is the PID adjustment, very crude. When I have time and desire I will try to improve it.

I consider cruise an emergency mode (if throttle is not installed).
In the event of a breakage of the clutch bearing (sprocket side) or of the pedal axle (it happens), with appropriate settings and by holding down the walk assist button, you can go home.
The settings, which I do not set as default because it becomes a throttle and therefore illegal in the EU, are these:
"Speed cruise enabled" = 0
"Cruise without pedaling" = enabled
possibly "Cruise on street mode" = enabled
The speed is the one chosen with the assist level.

I add, in case you haven't read the manual, copy/paste regarding cruise:

_13 – Cruise mode
“Cruise assist” is a speed control assist mode.
- Speed level 1 - ECO
- Speed level 2 - TOUR
- Speed level 3 - SPORT
- Speed level 4 – TURBO
The value set in the levels is the target speed to be maintained in km / h or mph.
The power supplied by the motor adjusts itself to maintain the chosen speed.
Speed may not be reached due to limited motor power.
The speed limits seen above have priority.
Carefully read the function of the next parameter.
- Cruise without pedaling
Disabled.
The cruise activation mode is subordinated to the movement of the pedals.
Speed is maintained only with minimal pedal movement.
By stopping pedaling the motor stops. It is the default mode.
It can be compared to the “Cadence assist” mode, the difference is that changing the level does not change 
the power supplied by the motor but the speed to be achieved.
Enabled.
You can maintain your speed even without turning the pedals by pressing the walk assist button.
By releasing the button and stopping pedaling, the motor stops.
It can be compared to an accelerator with speed variation in steps. Increasing the level increases the speed,
decreasing it decreases, at 0-OFF the motor stops.
Available only with brake sensors installed and enabled.
Inquire about legislative restrictions regarding use.
In STREET mode with cruise mode disabled, assistance is activated in “Cadence assist” mode.
- Speed cruise enabled
Minimum speed limit for activating cruise mode, in km / h.
Below this value, assistance in “Cadence assist” mode is active._

dzid26, 2025-02-25:
> I don't know what Bafang is like, but I think maintaining speed without holding down a button or throttle is very dangerous.

Cars do it all the time. The trick is to allow it above certain minimum speed. 
(Here also a minimum time after non-zero speed due to filter occasional fake speed when reversing).   

I could imagine such optional feature to be always active regardless the assist mode. 
1. Hold throttle for torque
2. Release throttle (or walk assist) while pedaling engages cruise 
3. Releasing throttle (or any user input) while not pedaling disengages cruise