emmebrusa/TSDZ2-Smart-EBike-1 #84: Step #3 on wiki homepage out of date?

fusenuk, 2023-10-04:
I'm new to this and was reading through the wiki steps and #3 appears to be out of date -

_download and install [ST Visual Development](http://www.st.com/en/development-tools/stvd-stm8.html). Install it directly to C:\ , not to the default folder. The file STVP_CmdLine.exe must be located in C:\STMicroelectronics\st_toolset\stvp. Alternatively you can set the correct path in the file _TSDZ2-Smart-EBike/src/controller/compile_and_flash.bat_

The file _compile_and_flash.bat_ has now been separated into _compile.sh_ and _flash.sh_, so the path is now set within _flash.sh_, but also, I installed STVD on Windows 10 using the default path it suggested and the flash.sh file already includes the path it used -

PATH = %PATH%;C:\STMicroelectronics\st_toolset\stvp;"**C:\Program Files (x86)\STMicroelectronics\st_toolset\stvp**"

So it seems like step 3 doesn't need to include the parts about changing the install path? I was going to edit the wiki page myself but thought I'd ask in case I've missed something obvious to why it shouldn't be changed.

dzid26, 2023-10-05:
It seems so. You can edit the wiki further if you wish.

fusenuk, 2023-10-05:
I've slightly modified it after your recent edit. I'll mark this as closed, thanks!