emmebrusa/TSDZ2-Smart-EBike-1 #85: Missing date on generated files and not sure if doing it right

jonasborneland, 2023-12-07:
Hi

First of all thanks a lot for this awesome project.

I'm not sure if I'm doing this right. But the date is missing on the created files.

Example: "TSDZ2-20.1C.2-2-.hex" ... Should have been someting like: "TSDZ2-20.1C.2-2-12-07-2022.hex" I guess.

I'm using macos.

Doing the following:

In the root folder of TSDZ2-Smart-EBike-1

Running this in terminal:
`open JavaConfigurator.jar `
then configuring my settings and clicking Compile & Flash
Then the only thing seems to be happening according to latest modified date is a new experimental settings file is created - nothing else changes – is this correct? or should there be created more files?
Then i run:
`sh compile_and_flash_20.sh`
and the files compiles and creates files in bin folder and releases folder
and it successful transfers to the bike ending in a all completed.
But I'm not sure if the files created is based on my actual settings or just some default values?

I really hope you can help me out, and thanks a lot! :-)

dzid26, 2023-12-08:
> Then the only thing seems to be happening according to latest modified date is a new experimental settings file is created - nothing else changes 

Compiled files in /bin folder are not created ?

Regarding the date, I am on Windows, and it's also not being added to the file name. Not sure why, but it seems like a minor issue. Feel free to investigate:
https://github.com/emmebrusa/TSDZ2-Smart-EBike-1/blob/93f874bc267d116a0f95faedf1d8cfb5b2ffc4d8/tools/Java_Configurator_Source/src/TSDZ2_Configurator.java#L1298-L1304
(I think `backup_name` becomes `$1` parameter in the batch script.)

jonasborneland, 2023-12-08:
Thank you so much for the answer

Yes while clicking the clicking Compile & Flash only `experimental settings file` is created like `20231207-222908CET.ini`

But afterwards when running `sh compile_and_flash_20.sh` all files are created in `/bin` and `src/controller/config.h` gets modified.

But my worries is if `sh compile_and_flash_20.sh` does modify/compile the files based on my settings? or if the Compile & Flash button in the java should do more than just the `20231207-222908CET.ini` file? :-)

dzid26, 2023-12-09:
`Ini` is files store settings for the Java interface. When you click Compile & Flash,  Java configurator modifies also `src/controller/config.h` which contains bunch of switches and parameters for the rest of the bike control software. The software gets compiled into bin folder and at the end of the process `main.ihx` is created ther.  `main.ihx` is actually the full software and is flashed to the bike.  
It is also renamed and copied to the releases, but it's not useful there to most people, as it's easy to simply compile again if needed. 
For some reason your release file name doesn't contain the date, but it doesn't matter. 

I am pretty sure you are getting your software correctly flashed with all the changes.

jonasborneland, 2023-12-09:
Thank you very much. I was not aware of `src/controller/config.h`
Compile & Flash in the Java configurator actually modifies the above file, so I'm more sure not that it work as supposed to :-)

I found a quick fix for the date.

In `compile_and_flash_20.sh`

I changed the line `settings_date=$1`

to `settings_date=$(date +%F)`

Thank you for the help