emmebrusa/TSDZ2-Smart-EBike-1 #87: coaster brake function implementation : will damage the controller if not physically present ?

aston89, 2024-01-13:
1) Why i ask this question : 
After just 3 months of use the sprag-clutch (inner side) of the tsdz2b is starting to fail.
Replacing only the sprag-clutch is not possible, it's embodied into the big gear.
The replacement shipped to europe will cost something like 70 bucks or even more.
Old tsdz2b had the sprag-clutch removable, replacement would had cost 15 or 20 bucks.
I use the bike every day and do a lot of km.
I am not willing to spend a fortune on spare parts for something clearly bad designed and incline to fail so much.

2) What i want to do to eradicate the problem : 
Opening the sprag-clutch seals and cementing it using metal epoxy, basically transforming the tsdz2b into the coaster brake counterpart (it did existed in the tsdz2 era).

3) What i need to know before doin this : 
Since the backpedal function is not physically present i'll be able to backpedal infinitely if i need.
Due to the fact that the sprag clutch will be locked, the motor will "back spin" with the pedals (will not happen pedaling normally fyi) so while backpedaling, even for a short period of time (ex. chain lubing, positioning the pedal to start at redlight etc etc) an unknown amount of current will flow to the controller like if it's a back EMF (but i could be wrong about this) 

4) questions finally !

A: Will the current generated by backward pedaling damage the controller ?

B: The "coaster brake" option in the configurator will override the "ramp-down" (deceleration) of the motor making it stop istantly ? (Is it maybe comparable to "ramp-down=100" ?) 

C: The "coaster brake" option in the configurator will also change the "ramp-up" value if pedaling again after backpedaling ? 

D: would you recomend me to enable the coaster-brake option in the configuration even if i dont physically have it for safety reason ? ( the actual ramp-up is 20 while the ramp-down is 100)  

thanks in advance for every usefull information you could provide.

dzid26, 2024-01-13:
> After just 3 months of use the sprag-clutch (inner side) of the tsdz2b is starting to fail.
> Replacing only the sprag-clutch is not possible, it's embodied into the big gear.

How do you know it is this clutch? [What is the symptom](https://endless-sphere.com/sphere/threads/tsdz2-motor-slipping.118921/post-1747243)?  Is the clutch slipping due to motor torque or due to your legs force? 

Btw, even in normal Tsdz2, motor spins backwards if bike is rolled backward and chain rotates backward. 

A: Can you generate too much current?  Let's see. So when mosfets are opened current still can flow to the battery via mosfets' body diodes. Uncontrolled recuperation is not very efficient and I think it needs high rpm to achieve high current.  Assuming it does what would be damaging current (which would correspond to backward torque needed).  K80E08K3 forward voltage drop is 1.5V. Thermal resistance of junction to heatsink is 0.75C/W.  Max temperature is 175C, but let's say 100C.  So 100/0.75/1.5=88A.  So instantaneous temperature is not a limitation. So then just all the normal average current limits of what case can dissipate. Since we are talking about you pedaling backward you may generate maybe 1A at high cadence and virtually no temperature rise. 

E: Backward pedaling result is 0 cadence in the software. https://github.com/emmebrusa/TSDZ2-Smart-EBike-1/blob/93f874bc267d116a0f95faedf1d8cfb5b2ffc4d8/src/controller/motor.c#L853-L856

aston89, 2024-01-14:
i know it's the clutch (the little one of the two present in the big gear of tsdz2b) because when i put too much pressure it makes like an harsh sound and slip (pedals goes free until i rewind them, no possibility to move the chainring using pedals)

ok so i should not worry about generating accidentally a spike voltage while backpedaling , that's good !

closed 2024-01-25 by aston89