emmebrusa/TSDZ2-Smart-EBike-1 #88: Coaster brake torque threshold, how it work ?

aston89, 2024-01-25:
hello there, converted my unit to a coaster brake and enabled the option in the configurator but i have encountered some problems.

It's seems like the power-assist-mode is programmed in a way that gives an omogeneous and fluid assistance even if the torque drop a bit while "switching" from a leg to another, this delay also looks like a necessary time window for calculating the human power, this apparently make the ramp-down useless even if set to 100.

This issue cause a "bump feel" if you istantly stop pedaling that apparently looks like a "rotor inertia" but it's not, tested with multiple level of assistance at same cadence and the bump become stronger, i need to throttle down the cadence a bit before stop, this happens to be dangerous in some circumstances, the motor torque can lift me up very easily and i am 120kg.

The coaster brake option when enabled seems like to help but feels "random", sometimes the "delay" remain unaffected and i tried to test this all day long but wasnt able to replicate the same behaviour for long expecially at slow and high cadence. 

So in order to solve this issue witouth randomly trying everything i need to ask : 

1) Could it be the cadence sensor and how it's designed ? not precise enough ? or maybe it's the way how the cadence value is updated in memory ? (like after a fixed number clock-cycles // like a round robin system wheere it happens when every other task is completed, wich could cause unexpected delays if a task require too much time )

2) Is the torque-assist-mode quicker compared to the power-assist-mode ? (theorically it should not require to wait a bit for the calculation of the human power)

2) the parameter "Coaster brake torque threshold" how does it work ?
It can be set from 0 to 255(s), what does it do if i put to 255 ?
Will it override every other "ramp" value ? (even the sudden ones integrated in the assistance modes ?)

jussisaily, 2024-01-25:
Hi, check my old closed thread with the recommended coaster brake settings. The pedal bump when stopping pedaling is not totally gone but not so bad with these.

https://github.com/emmebrusa/TSDZ2-Smart-EBike-1/issues/36

dzid26, 2024-01-25:
> It's seems like the power-assist-mode is programmed in a way that gives an omogeneous and fluid assistance even if the torque drop a bit while "switching" from a leg to another, this delay also [...]

Also try changes from #63

aston89, 2024-01-25:
> > It's seems like the power-assist-mode is programmed in a way that gives an omogeneous and fluid assistance even if the torque drop a bit while "switching" from a leg to another, this delay also looks like a necessary time window for calculating the human power, this apparently make the ramp-down useless even if set to 100.
> > This issue cause a "bump feel" if you istantly stop pedaling that apparently looks like a "rotor inertia" but it's not, tested with multiple level of assistance at same cadence and the bump become stronger, i need to throttle down the cadence a bit before stop, this happens to be dangerous in some circumstances, the motor torque can lift me up very easily and i am 120kg.
> 
> Also you can try changes from #63

Nice, finished now to read that thread and it seems that you managed to make the motor extremely responsive witouth the so called stop-delay  but at the same time with a gentle startup, am i right ? 
So you achieved this goal bypassing the filters and other parameter being configurable in the configurator ? 
This feature called "smooth startup" became an assist mode in the configurator or it got integrated as an option ?
Do i have to intgrate this code in the one i already have ?

dzid26, 2024-01-25:
> Update 1 : after further investigations on the https://github.com/emmebrusa/TSDZ2-Smart-EBike-1/pull/73 provided by @dzid26, it seems that also enable Assistance-witouth-pedaling but since i dont use brake sensors im not truly incline to test it (even if i want i dont know how to compile/merge or whatever is needed to use this feature).
Im using the vlcd5 so probably if i got it right, it's a feature usable only with non-stock displays ?

I have vlcd5 too. It specifically doesn't require brake sensors. It's much smoother to start and also I tried to remove as much delay as possible.  The only thing is, if you have 36V motor, emmebrusa [reported ](https://github.com/emmebrusa/TSDZ2-Smart-EBike-1/pull/73#issuecomment-1826234849)some issues on his, possibly. 

You could try setting `CADENCE_SENSOR_STANDARD_MODE_SCHMITT_TRIGGER_THRESHOLD` to 0. 

If you want to try the full change, you can just download the whole thing from my [branch](https://github.com/dzid26/TSDZ2-Smart-EBike/tree/ebike_work) which tracks that PR - and use it as usual.

closed 2024-05-15 by aston89