emmebrusa/TSDZ2-Smart-EBike-1 #91: Limit power assist mode to 250W?

kiwisibk, 2024-04-10:
Hello everyone,

if you have a 48V 750W motor, is it possible to limit the power assist mode to 250W? So that you are legally OK in the EU with max. 250W and 25kmh?

Many thanks for your feedback!

dzid26, 2024-04-10:
Have you tried reading the [manual](https://github.com/emmebrusa/TSDZ2-Smart-EBike-1/blob/master/manuals/EN-Parameter_configurator_guide-TSDZ2-v20.1C.2-2.pdf)?
[image]

kiwisibk, 2024-04-10:
Street mode is clear, I want the 250W in power assist mode. But I think I have understood that you set it in the "Battery Power Max" field.

dzid26, 2024-04-10:
If you put 250W Street power limit it will also apply to Power assist.  

Street / offroad modes are not separate assist modes.  Those limits are applied on top of assist modes.  

You can ride with Power Assist on the street (where legal limits would apply) and offroad (where you don't care).  It is in Street mode by default. https://github.com/emmebrusa/TSDZ2-Smart-EBike-1/blob/master/manuals/EN-Operating_manual_of%20_display-TSDZ2-v20.1C.2.pdf

kiwisibk, 2024-04-11:
Great, thank you.

Can you perhaps explain to me the button sequence on the VLCD6 to disable
street mode if it is set as default at startup?

Thanks again

KILLERD0G, 2024-08-13:
MY 250w 36v settings that work well.
[image]
[image]
[image]