emmebrusa/TSDZ2-Smart-EBike-1 #93: Parse/write directly to src/controller/config.h instead of ini

dzid26, 2024-04-28:
[ini files](https://github.com/emmebrusa/TSDZ2-Smart-EBike-1/blob/master/experimental%20settings/20220410-135823CEST.ini) are meaningless. 

I would appreciate more if the [src/controller/config.c](https://github.com/emmebrusa/TSDZ2-Smart-EBike-1/blob/master/src/controller/config.h) was written to and parsed directly. 

The advantages:
- config.h would be reflected in GUI
- easier reading and trying other's people settings
- less chance for a mistake of flashing wrong settings

The backups and proven settings of config.h could be stored as usual. 
Also, settings should be stored together with the backed up binary in [\releases\backup](https://github.com/emmebrusa/TSDZ2-Smart-EBike-1/blob/cb8b0321e2931d206b54e83773fb469b06325235/compile_and_flash_20.bat#L17)

ec-bikes, 2024-05-05:
It seems like part of the purpose with ini files was to allow saving/loading multiple sets of settings, though I guess you could do the same with C header files (the GUI would just need to copy the right one into place before flashing). Another idea would be saving the settings in a different format such as JSON which at least includes names with the values.

dzid26, 2024-05-05:
Hmm,
- The benefit over json is that the backup is simply moving a file, so the parser/writer can't screw it up. Json would be useful if we were to modify other parts of the firmare, but historically only config.h was touched.  
- The config.h is already very simply structured.
- The header would be a single source of truth for the current setup state.

emmebrusa, 2024-05-07:
Yes, it would have been better to have a single file, but unfortunately there are two.
Changing now is complicated and I don't see much benefit for the user.
Also we should maintain compatibility with current .ini files, don't you think?

For those who want to contribute, there is a change that I think is more useful, instead of using File Explorer to move the files from the "experimental settings" folder to the "proven settings" folder and renaming them, be able to do it directly in the configurator.
It's on the list of things to do, the last one.