emmebrusa/TSDZ2-Smart-EBike-1 #99: E07 overcurrent shows up often

dzid26, 2024-06-13:
Sometimes it enters E07 when pedaling and stays there until restart.

I also notice I get less maximum torque using the same settings as previous software. I am not sure if that is related.

There is fairly large amount of complexity around Battery maximum power. But this shouldn't even exist. Max Battery power is a marketing term and not physical limitation. Current is always the limit. (And why would we allow more current at lower voltage??). 

If anything, battery max power could be converted to another battery current limit only _once_ - based on rated battery voltage (cells x 3.6V) - not startup voltage readings.  

I am not sure if the problem is there, but it's likely.  If I have a high battery voltage after charging, the limit will be low.  That would explain why at the end of the ride I was not getting the error anymore.

emmebrusa, 2024-06-14:
The E7 overcurrent error has been present since version 4-update-3 for 860C, and has now been added to the stock display version as well.
Overcurrent control is important.
In some versions of OSF there were cases of burnt mosfets, one hypothesis was a bug breaking the motor control loop, so a current control was added in the main loop too.
During one ride I had an accident, on a very steep climb, I made a risky gear change, the chain fell, got stuck and broke.
The motor was at maximum power and consequently the blue gear was also damaged, but the E7 error appeared and this was enlightening. I understood that the current had reached 23A (overcurrent limit at the time, the battery limit was 17A) and that this control could also be used to protect the blue gear.
The E7 control time was every 100 ms, so I reduced it to 25 ms.
Subsequently the same accident happened to me, the chain fell and got stuck, it didn't break, the motor stopped due to the intervention of E7. Turned it off and on again, no damage to the blue gear.
I also had confirmation of this from another episode, on a path there was a very high root, I should have got off the bike, but instead I pushed hard on the pedals to overcome it, I got stuck. E7 intervened and I got out without harm.
How can an overcurrent occur? Is the question I asked myself. The answer is simple, motor control does not limit the current, the rule! If the acceleration is very fast, at maximum power the limit is exceeded.
Only after exceeding it does the current decrease, following the deceleration ramp, until it stabilizes at the maximum limit.
While with the intervention of E7 the interruption of the power is immediate and it is then necessary to turn the display off and on again.
The current peak may be false, due to noise on the ADC input.
To have immediate intervention, the instantaneous value of the ADC is controlled without filters. Applying a software filter would decrease the effectiveness of the function.
Now the E7 trip point is 8A above the maximum battery current (in the 860C version it is 6A).
It's in main.h, #define ADC_10_BIT_BATTERY_EXTRACURRENT 50.
I increased it after a report like yours of unwanted interventions. However, the user did not solve the problem by increasing the limit, he instead solved it by decreasing the acceleration ramp.
See if it works for you too.
Otherwise we need to understand the cause.

emmebrusa, 2024-06-18:
Have you had any other unwanted E07 errors?
I forgot an important detail, in this version E07 is a shared error code.
It could be BATTER OVERCURRENT, but also MOTOR CHECK (IF GOES ALONE), it could also be a long inertia of the engine with deceleration ramp at 100%, possibly try increasing MOTOR_CHECK_TIME_GOES_ALONE_TRESHOLD (ebike_app.c).

dzid26, 2024-06-18:
Oh, it's not good to reuse values like this.  I don't see immediate reason why `ERROR_MOTOR_CHECK` couldn't be checked separately... 

I will eventually figure out the root cause of the issue and let you know. 
First I am working on porting my other changes to the new version.

emmebrusa, 2024-06-19:
Yes sharing an error code is not a good idea but I have to do it.
The reason is simple, I can't know which codes are available on all displays.
For this reason I prefer to limit the use to the 9 foreseen by the Tonsheng protocol.
The first idea was to share E09 ERROR_WRITE_EEPROM, the chances of getting this error are zero.
Then for a stupid reason that doesn't need to be explained, I opted for E07, also based on the belief that ERROR_MOTOR_CHECK is now unlikely to occur.
But a false error is possible, the control time of 4 seconds must certainly be increased.
And anyway I already changed ERROR_MOTOR_CHECK from E07 to E09.
The previous error code E05 is used for ERROR_THROTTLE, as in version 860C.

dzid26, 2024-06-19:
Just from the _good practices_ point of view the check should be explicit:
`(ui8_system_state == ERROR_BATTERY_OVERCURRENT) || (ui8_system_state == ERROR_MOTOR_CHECK)` 
even if underlying values were the same and even if sdcc fails to optimize.

But anyway: `ERROR_MOTOR_CHECK` = E09 seems like a good idea.  

Btw, although it doesn't matter, I think it would be more logical if `ui8_motor_check_time_goes_alone` was initialized with 0 and counting up:
https://github.com/emmebrusa/TSDZ2-Smart-EBike-1/blob/0e553a59e1603362b1027080e46f49cf012721d0/src/ebike_app.c#L1677-L1687

and then:
```
if (ui8_motor_check_time_goes_alone > MOTOR_CHECK_TIME_GOES_ALONE_TRESHOLD) {
	ui8_system_state = ERROR_MOTOR_CHECK;
}
```

---------

I think 4s is plenty time for the motor to stop, so I don't think this is the culprit of my issue.

emmebrusa, 2024-06-27:
I'm worried about E07 overcurrent.
I need to understand if it is a real error or a false error.
After moving ERROR_MOTOR_CHECK to E09, I got a report from another user.
He tried increasing the overcurrent threshold and decreasing the motor acceleration, but it didn't solve the problem.
Maybe it could be a slow mosfet?
Try setting #define PWM_FREQ 18 in main.h?

emmebrusa, 2024-06-27:
Update to the latest comment.
There was a misunderstanding with the user who tried, he has now confirmed to me that the error is E09 ERROR_MOTOR_CHECK.
So, if the problem is the same for you, try increasing the threshold time (60 or 80).

dzid26, 2024-06-27:
I am not getting the error anymore. In the meantime I did some software [customizations](https://github.com/dzid26/TSDZ2-Smart-EBike/commits/experimental/) and further changed settings.  I also installed new battery pack :) 

Last few rides I used was with the following settings and there was no error:
[image]
[20240626-144324WEST.txt](https://github.com/user-attachments/files/16019719/20240626-144324WEST.txt)

I dug out a two configurations which I think can give me some clues. 
First one I believe was when I was getting error 07 several times and second when I think it the error didn't appear anymore. At the time I thought what helped was setting the "battery power max" to 800W. But it could be anything.... 
[20240612-201118WEST.txt](https://github.com/user-attachments/files/16019629/20240612-201118WEST.txt) ini
[20240617-164923WEST.txt](https://github.com/user-attachments/files/16019628/20240617-164923WEST.txt) ini
[image]
[image]

On my next ride, I will try that offending setting again to see if the error reappears.

dzid26, 2024-06-28:
I tried the last known bad setting and i didnt get the error. I probably should try again, but this time without any of my code customizations.

I don't believe in "slow mosfet" or noisy ADC theory (the margin is 8amps (`ADC_10_BIT_BATTERY_EXTRACURRENT`).

I have some "PWM theory". 
Imagine the motor is rotating at base speed. 
[image]

The PWM is maxed out at 255.  If you slow down motor faster than pwm ramps down then the produced current might be higher than current limit. This could possibly happen while pedaling approaching a hill or braking.   Would that happen in practise? I don't know. Above [70rpm cadence](https://github.com/emmebrusa/TSDZ2-Smart-EBike-1/blob/0e553a59e1603362b1027080e46f49cf012721d0/src/ebike_app.c#L629-L633) the deceleration is set to maximum, minimizing the chance of that happening.  But maybe there is some other scenario when PWM is temporarily higher than it should be and high current is generated.

I think few times when the error happened for me was around the speed limit (25kph) when the motor stops and starts often and I was at high cadence. This could be a clue, but I don't remember if that was the only times.  

For some reason, the speed limit was not working with my customizations.  I need to figure that one first.

emmebrusa, 2024-06-30:
> I tried the last known bad setting and i didnt get the error. I probably should try again, but this time without any of my code customizations.
> 
> I don't believe in "slow mosfet" or noisy ADC theory (the margin is 8amps (`ADC_10_BIT_BATTERY_EXTRACURRENT`).
> 
I don't think so either, but keep in mind that the adc value is not filtered.
With the 860C version I had 2 reports of E07 overcurrent in addition to mine (there is no motor check on this version).
The first from a friend, who obtained it with acceleration at 50% and with a max current of 20A, the overcurrent limit then was 23A (copied from mspider65).
I then had confirmation that with acceleration at 35% it didn't have the error, I still decided to change the limit, to max current + 6A (which for my friend is 26A).
The second report, from an Endless user, I advised to try to decrease the acceleration or increase the overcurrent limit, but I no longer received any answers.

In the stock display version, at the first report of E07, I decided to change for max current + 8A.
Also because I saw that mspider65 in its latest version has decreased the max current to 16A and increased the overcurrent limit to 25A.

The reports of E07 received for the stock display version, I think are all caused by "error motor check", including yours.
I realized that it is not enough to increase the control time, I also have to compare a minimum number of erps, not zero.
>
> I have some "PWM theory". Imagine the motor is rotating at base speed. 
> 
> The PWM is maxed out at 255. If you slow down motor faster than pwm ramps down then the produced current might be higher than current limit. 
> 
Yes, I also think that if the motor slows down faster than the PWM, the current increases.

There are some strange settings in your screenshots...

dzid26, 2024-07-03:
I was able to reproduce E07. 

I have an impression it was happening only around speed limit. Can others confirm?

emmebrusa, 2024-07-05:
We need to have confirmation whether the error is E07 or E09.
[https://github.com/emmebrusa/TSDZ2-Smart-EBike-1/commit/81b813c0b0272b292ac54c3a56346855758ba5f1](url)

PR #104 (2024-08-09): Cut PWM above speed limit (fixes E09)
I was suspecting the [issue](https://github.com/emmebrusa/TSDZ2-Smart-EBike-1/issues/99) was E09 instead of E07 lately. After using your last change https://github.com/emmebrusa/TSDZ2-Smart-EBike-1/commit/81b813c0b0272b292ac54c3a56346855758ba5f1 I was indeed getting the E09 error. 
 

After adding this modification, E07/E09 errors didn't appear. 

The change should be done regardless of the errors anyway.

- This is a simple PR that quickly Fixes #99. I have more improvements to this function coming in a next PR. 

### Rationale:
Speed limit function doesn [... cut]
closed 2024-08-09 by emmebrusa