emmebrusa/TSDZ2-Smart-EBike-1 #100: Battery current display doesn't work

dzid26, 2024-06-16:
This was also an issue for me for previous software. 
For some reason the current doesn't display with "lights On".  Currently it just shows up wheel speed instead. 
Some other data, like cadence shows up ok.

So far I couldn't figure out from code what is wrong.

emmebrusa, 2024-06-17:
Strange. I tried it and it's fine.
However, I use XH18 which has an immediate display of the value.
If you use VLCD5 the value increases slowly.
Maybe the viewing time is too short?
Try setting:
Data 1 = 3 (battery current)
Time to displayed data 1 = 0 (no time)
Number of data displayed at lights on = 1
If it doesn't display the current battery, post the .ini file which I check.

dzid26, 2024-06-17:
Thanks for checking that. 

> Time to displayed data 1 = 0 (no time)

Ok this makes it show the value - zero...  
But that was rookie mistake. I got confused because I currently run on Level 1 Torque assist (40) - because my battery is dying- and the current doesn't exceed 4Amps usually.  Since VLCD5 starts to show values from 3.5kph, I kept seeing 0.

closed 2024-06-17 by dzid26