emmebrusa/TSDZ2-Smart-EBike-1 #102: TSDZ8

ShalokShalom, 2024-06-28:
Is there any interest in adding support?

I consider buying one, and could assist with executing code, testing and that like.

emmebrusa, 2024-07-03:
I don't think I will also deal with TSDZ8, it has a different microprocessor, probably the same as the TSDZ2 v2 controller.

KILLERD0G, 2024-08-13:
Thats a real shame becasue my friend bought one of these motors and said its soooo much better than the tsdz2 or b. Looks like this will be the motoro to grab in the future...

dzid26, 2024-08-13:
"soooo much better" means nothing.  What parameter is better exactly?

KILLERD0G, 2024-08-13:
Well for a start my friend has tried the tsdz2 and then the tsdz2b and he says the tsdz8 pulls better and is faster. He has the 500w version . In the UK you can only have 250w at 15.5 mph assist... All i know is he prefers the new tsdz8 version as it pulls better..

closed 2024-08-14 by ShalokShalom

ShalokShalom, 2024-11-08:
> I don't think I will also deal with TSDZ8, it has a different microprocessor, probably the same as the TSDZ2 v2 controller.

Just for information: There is an updated firmware for the 8, and an official recommendation for an update kit and SDK: 

https://youtu.be/dN-aB0hSGLc?t=2m20s