emmebrusa/TSDZ2-Smart-EBike-1 #110: TSDZ2 battery consumption

getenvy89, 2024-07-17:
Hello! I used your firmware to improve engine performance, and I have a question about the firmware parameters. I mainly use assistant levels 2 and 3 on the stock engine and differentiated the levels in the firmware similarly. There is a problem with energy consumption; it is approximately the same at levels 2 and 3, even though level 3 is more powerful. What could be the issue with the parameter settings?
[image]
[image]

getenvy89, 2024-07-20:
I'm using eMTB mode. Battery consumption with my settings at levels 2 - 3 - 4 is approximately the same. My battery is 14.4ah and lasts for about 70 km. The speed is no more than 35 km/h, engine no longer moves.

kiwisibk, 2024-07-25:
I would also be interested to know what parameters you can use to get the best possible range. I also only have a 12.5 Ah battery and a 48V 750W motor. Ideally, I would like the best possible range configuration with the option of switching to power mode for a short time if necessary. Thank you