emmebrusa/TSDZ2-Smart-EBike-1 #111: Battery setting for 48v motor 500w with 36v battery?

jaybeemlanuza, 2024-07-30:
I'm getting a 48v motor but only have 36v batteries, can this still work with the right battery setting in the configuration?

dzid26, 2024-07-30:
It should work, but you will have no assist above ~65rpm cadence, which is quite low.

KILLERD0G, 2024-08-13:
If your getting a 48v motor then to get the most out of it you need 48v batts...

dzid26, 2024-08-13:
Actually, I would recommend at least 14s3p to have a decent power and high cadence.  

https://endless-sphere.com/sphere/threads/tsdz2-with-15s-55v-nominal-63v-max.119804/
Recently I came to the conclusion 15s charged to about 60V or 16s charged to 63v would be even better. (Note capacitors are rated 63V). 
The reason is the voltage initially drops quite quickly and you end up being at lower voltage most of the time which even with 14s is very noticeable and not great for max power and max cadence.  

Field weakening is not as needed with higher voltage. 
Another benefit of 15s charged to a lower voltage is that it adds to longevity of the battery.  

One caveat is that the BMS must allow for balancing at any voltage, not just close to 100% SOC.