emmebrusa/TSDZ2-Smart-EBike-1 #112: Flashing Issue, SWIM connection?

UteAkaKute, 2024-08-11:
Hey everyone! 

I'm trying to flash using a 36V TSDZ2 with XH18 display. The program seems to compile just fine but I am not able to get it to flash.

[image]

I have checked the SWIM connections which seem to be just fine. I doubt this is actually a hardware issue since I have been able to back up the stock firmware using STVP by reading the program memory, data memory, and option memory. STVP sometimes also gives this error, but I am able to read it after one more try. 

[image]

I have been power cycling while pulling the RST pin low on my ST-LINK V2 using the GND pin. 
I have updated the firmware on my STLINK v2 using STM32CubeProgrammer
I have checked the main.hex (which seemed fine and not corrupted) and I have checked data_empty.hex, which was not in .../bin. ? 
I have checked the STlink connections numerous times. 

Does anyone know what I can do to solve this flashing issue?

dzid26, 2024-08-11:
Nothing to do with the TSDZ2 firmware.

Things to try:
- Disconnect bike charger when flashing
- Turn off the ebike battery, unpower USB - this will force STM8 to be unpowered and makes sure stm will reset
- Try flashing [optionbyte](https://github.com/emmebrusa/TSDZ2-Smart-EBike-1/blob/master/src/option_stock.hex) with stvp.
- Try changing stm8 programming cable power pin from 5V to 3.3V and then flash.  (Bike technically should be always kept off when you do that, but from my experience the programmer doesn't die even when 5V gets to 3.3V line when ebike is on. )

UteAkaKute, 2024-08-11:
@dzid26 Wow thanks for the quick reply!

- Bike charger is not connected while trying to flash
- Turning off the ebike battery and unpowering the USB does not make a difference, I have also tried this numerous times when power cycling
- switching to 3.3V also does not make a difference, I have tried this before. 

- I was able to flash [optionbyte](https://github.com/emmebrusa/TSDZ2-Smart-EBike-1/blob/master/src/option_stock.hex) This also did not change things. I did see this optionbyte was different from the option byte I backed up: 

`S104480000B3
S104480100B2
S10448032888
S104480500AE
S104480700AC
S104480900AA
S104480b00A8
S104480d00A6
S104487e0035
S9030000FC`

dzid26, 2024-08-11:
> Turning off the ebike battery and unpowering the USB does not make a difference, I have also tried this numerous times when power cycling

Looking at my old note https://github.com/vdudouyt/stm8flash/issues/123#issuecomment-2053753930 I have one thing to add. - When you power down the controller, wait 5minutes for the capacitors to discharge. 
- I remember I thought my controller was dead, because it stopped responding to flashing. In reality, I think due to option byte corruption it was just not resetting. 

Other than that, it could be simply broken contact.

UteAkaKute, 2024-08-11:
I have tried this waiting for capacitor discharge a number of times now, also to no avail. 
I remember now I might have programmed [optionbyte](https://github.com/emmebrusa/TSDZ2-Smart-EBike-1/blob/master/src/option_stock.hex) using STM8S105x6 instead of STM8S105x4 which I was supposed to do I think. 
Now for some reason I can't read or program at all using STVP. The bike itself does still work. 
I'm ordering a new and better quality STLink somewhere to see if the connection issue might be due to a bad stlink. I read this here: https://www.pedelecs.co.uk/forum/threads/tsdz2-connection-error-with-st-linkv2.41873/

Wouldnt a broken contact be weird considering I was able to read and program?

dzid26, 2024-08-11:
When you wait for capacitors to be discharged, you can try powering up with nrst tied to ground as I mentioned in my [note ](https://github.com/vdudouyt/stm8flash/issues/123#issuecomment-2053753930) and then trying to flash option byte or unlock the chip with stm8flash.  

Bad contacts could work sometimes.

laurentholzman, 2024-12-09:
Hi @UteAkaKute, did the new STLink solve your pb ? I am experiencing the same issues right now. Everything works fine in STVP (read and program) but I am unable to flash with the javaconfigurator ("cannot communicate with the device")

UteAkaKute, 2024-12-09:
> Hi @UteAkaKute, did the new STLink solve your pb ? I am experiencing the same issues right now. Everything works fine in STVP (read and program) but I am unable to flash with the javaconfigurator ("cannot communicate with the device")

Yes, I got my hands on a proper STLink (one of these https://www.digikey.nl/en/products/detail/stmicroelectronics/ST-LINK-V2/2214535 ) and it worked like a charm. I have read on other forums about people having trouble with cheap chinese knockoff STLinks, so that probably was the issue here as well.

laurentholzman, 2024-12-10:
Thank you for the tip !