emmebrusa/TSDZ2-Smart-EBike-1 #113: TSDZ2 250w 36v....

KILLERD0G, 2024-08-13:
Just wondering when i try the beta version and i use throttle and pull the brakes hard, why does the throttle not enage again. Once the brakes are pulled on the motor goes really slow and struggle to start again, until i turn the bike off and then back on. Any ideas ? My settings are below....
[image]
[image]
[image]

KILLERD0G, 2024-08-13:
Forgot to add my brakes are hydralic, i am not using any brakes sensors....

pepsonEL, 2024-08-26:
Hi
Did you resolvced this problem ?

KILLERD0G, 2024-08-26:
Yes and no. I went back to the april relase and stopped using the beta's... I have changed my settings now and i am having some good results.