emmebrusa/TSDZ2-Smart-EBike-1 #115: E07 error at high speed in street mode

ipp0, 2024-08-20:
I am running 20.1C.6-beta-0 on TSDZ2 with coaster brake and VLCD5. I only use the bike in street mode. When riding downhill, the assist is disabled correctly at around 25 km/h, but when speed reaches ~32 km/h, E07 is displayed and the motor is disabled until it’s powered off and on again. Is this expected behavior? Is there a setting that can affect this? It is quite annoying that I need to stop at the lowest point, restart, and only then I can have assist in the uphill part of my commute.

dzid26, 2024-08-20:
Pull latest version. It is already fixed.

ipp0, 2024-08-22:
I can verify that beta-1 seems to have fixed the issue. Thank you!

closed 2024-08-22 by ipp0